padsley/k600analyser #34: Mac

padsley, 2015-01-12:


KevinCWLi, 2015-01-12:
This sounds great to me :) (completely objectively)… don't know how much help I can be on such issues but I shall look into it sometime.

padsley, 2015-01-12:
...I think you just volunteered... :+1:

KevinCWLi, 2015-01-12:
Qué? No Hablo Ingles… I shall endeavour to do so when I'm back.

padsley, 2015-01-12:
Well, the problem appeared when I tried to compile things on Luna's laptop so I'm going to assign the issue to her. Then she can also test whether it's fixed.

padsley, 2015-01-13:
Right. So, from what I can tell, the architecture can be found by using something along the lines of:

ARCH := $(shell uname)
ifeq ($(ARCH),Linux)
#	@echo "Linux";
OS_DIR = linux
OSFLAGS = -DOS_LINUX -Dextname
#CFLAGS = -Wall -lrt
CFLAGS = -Wno-cpp -lrt
# add to compile in 32-bit mode
# OSFLAGS += -m32
LIBS = -lm -lz -lutil -lnsl -lpthread -lrt #-lgsl -lgslcblas #-lvme
endif

in the Makefile.

This will automatically detect 'Linux' as the architecture option.

I think that the equivalent for OSX is 'Darwin'. If someone could check this by running 'uname -s' on a Mac and posting the output here, that'd be awesome. :+1:

padsley, 2015-01-13:
OH MY GOD ALL OF THE GIANT TEXT WHAT IS GOING ON?

KevinCWLi, 2015-01-13:
From what I've seen with some other makefiles, this is the way to go and darwin is correct for mac (or as Lee calls it: "funny linux"). Will get back to you asap.

KevinCWLi, 2015-01-13:
So I'm guessing you are looking at the midas Makefile? They have a nice way of auto detecting the os etc.
I tried fiddling around with:
OSTYPE = darwin
OS_DIR = darwin
OSFLAGS = -DOS_LINUX -DOS_DARWIN -Dextname
For the OSFLAGS I kept the -DOS_LINUX flag because the MIDAS Makefile keeps it for Darwin. The terminal output I get is still the same, which is: 
make: *** No rule to make target `/Users/kevinchingweili/MIDAS/midas/darwin/lib/rmana.o', needed by `analyzer'.  Stop.

By the way, can Luna read this? If not, please ask her what mac OS X version she is using (please be mountain lion 10.8.5……)

padsley, 2015-01-13:
OK. That's a bit weird (that it keeps -DOS_LINUX).

1. Check in darwin/lib to see if there is a rmana.o
2. Luna can read this. I am unable to comment on the big puddy-tats of Apple, however. She's also filling in forms so she can get paid &c. so she has bigger concerns that a Makefile.
3. Maybe I should get someone to buy me a Mac so I can test the codes on that as well as on Linux. PLAN.

https://www.youtube.com/watch?v=38aDWDUjlOY

KevinCWLi, 2015-01-13:
Yes…. Join us… https://www.youtube.com/watch?v=yUpbOliTHJY

padsley, 2015-01-13:
If anyone is going to make references to quirky British comedies, it'll be me!

https://www.youtube.com/watch?v=lj2ZlY2nQ3s

In other, useful, news:

I think that the following should be added to the Makefile:

ifeq ($ARCH,Darwin)
OSFLAGS = -DOS_DARWIN -Dextname
LIBS = -lm -lz -lutil
endif

I'm going to make this change in the SetupDev branch because... that's one I have open and is the branch in which changes relating to how the code is setup should be make. If someone with a Mac could checkout that branch and then compile it, I'd be grapeful.

KevinCWLi, 2015-01-13:
Hey Phil, there is no rmana.o in darwin/lib… because I forgot that my MIDAS on the mac side does not finish installing properly… therefore the error comes up.

Luna the error I get when installing MIDAS (at the make install command) is:
"install: illegal option -- D"

This is because:
"The GNU version of split is different to the OS X version of split.
OS X is based on BSD Unix, and Linux (often referred to as GNU/Linux) uses the GNU toolset."

Are you successful in installing MIDAS? One possible workaround online is to change the - D option with - d but this is not guaranteed to work apparently. I am working on it now. Also what OS X version are you running please?

padsley, 2015-01-13:
Luna can get Midas to compile fine. At what point does your version fail?

'make' or 'make install'?

pellegri, 2015-01-13:
Sorry, here I am. My version is OS X Yosemite 10.10.1

KevinCWLi, 2015-01-13:
make install
I have changed the -D option to -d but then I see
"install: darwin/bin/webpaw exists but is not a directory"
……. I think this was the function that the -D option provided….

Luna you didn't have to change anything when installing MIDAS?

padsley, 2015-01-13:
Eh. Do you need to do 'make install'?

Try adding export MIDASSYS = /path/to/midas/directory to your .bash_profile.

padsley, 2015-01-13:
after having done 'make'.

KevinCWLi, 2015-01-13:
Yeah I've already tried that (It was the first thing I did I think), the only problem being that rmana.o is not present in $midas/darwin/lib…….

padsley, 2015-01-13:
Kevin and I have fixed this. For future reference:

rmana.o will only be created if ROOTSYS is defined at compile-time. I.e., you need to have sourced ROOT before you compile MIDAS.

padsley, 2015-04-27:
Closing issue. The MAC definition in the Makefile just references Linux. I.e. if OS_DARWIN -> set OS_LINUX.

It also isn't the cause of Funny Linux flaking out at the end of the analyser (see #35)

closed 2015-04-27 by padsley