padsley/k600analyser #38: TTree Fill timeout

padsley, 2015-01-30:
This is a problem with SetupDev with some input files and not others. At the t1->Fill(); line, the code is timing out or crashing.

I can't find the origin of this error but it seems to occur when there is ADC information being read in and not only TDC information. Need to check which branches are being declared and thus being filled.

For clarity: when I run with some old datafiles, all is good. With new datafiles, not so much.

padsley, 2015-01-30:
Also, this is on SetupDev. It might appear on other branches as well but, in that case, I haven't noticed it.

KevinCWLi, 2015-01-30:
My 5 cents: this sounds ODB related doesn't it? Just reminded me of that coz I read the other day about Retief telling you to change some things in the ODB...

padsley, 2015-01-30:
...that might explain quite a lot...

OK. I shall try dicking around with the odb and seeing where that comes out.

padsley, 2015-02-06:
So, when I run this with a file from PR194, it works fine. With a file from PR227, it doesn't work at all. Argh.

This suggests that maybe Kevin was correct and the odb might be the problem. Not sure whether I can get hold of an alternative.

padsley, 2015-02-06:
Sooooooooooooooooooooooo... this problem occurs but only sometimes. Argh. I just sorted one run from  PR227 (numero noine) and it got stuck. Run 53 worked fein. I think this might have something do to with empty data from the focal plane or something for runs sans useful data or summit.

Aussi, from the log book, the silicon calibration data start at number 81 and not before. I think that Lee might have been doing some DAQ testing for earlier runs.

So... I'm going to only sort the runs which are... y'know... actually useful. Maybe that fixes the problem. It might be invalid information being put in a branch which causes this?

padsley, 2015-04-27:
I'm really don't understand why I didn't close this before now...

closed 2015-04-27 by padsley