padsley/k600analyser #41: SetupDev -> Master Merge

padsley, 2015-04-13:
At some point in the not-too-far distant future, I want to merge SetupDev into the Master branch. Certainly before the coincidence experiments which are scheduled for later this year. (Mine I don't care so much about as I'm quite OK with using SetupDev).

I will warn people when this is about to change so that they are able to fork the Master branch if they are worried about their sort code not working afterwards.

KevinCWLi, 2015-04-13:
Cool beans, I shall definitely have a look at this in the future… I think for PR226 I am going to stick with what I am doing but for the future I shall start with it.

By the way, I have implemented the penetrability dependence of the fits etc….. So maybe we should add penetrabilities to the AngCorr list too sometime…….. I think that the 9.641 meV state in 12C has some of that asymmetry to it.

Kind regards
Kevin

padsley, 2015-11-04:
This has been done with the upgrades from @along4. Closing this issue.

closed 2015-11-04 by padsley