padsley/k600analyser #44: Ghost hits and 8Be???

padsley, 2015-04-26:
I've been thinking.

In order to look for 8Be, the user should ensure that they are looking for two hits on one silicon detector and that the two hits observed have relative energy of 92 keV.

However. In many cases, this can produce events within the detector which appear to have two alphas with similar energy in two front and two back strips and this can result in 'ghost hits', where two front hits and two back hits are converted into four hits in total.

I do not yet account for this in the main sort code.

I am also unsure as to how to deal with it. In principle, I'd like to do this in the SiliconData class by making a class which is able to deconstruct the TDC hits in each channel for the fronts and checking to see that no more than one hit is accepted per front channel. This would then require some reprocessing of the data to remove the ghost hits.

Alternatively, this could be hard-coded in the end of the MMM.c sort routine: a GhostBuster class which scans the event, tests it and fixes it if ghost hit conditions are found. This will cause the event to be voided and resorted with a more stringent algorithm or something like that. Or the original sort algorithm can be redone to take the possibility of ghost hits into account.

Anyway... I don't know what the best way of doing this is yet and that's a problem. I'm leaning towards changing the structure of the sort algorithm but I don't quite know yet. Any guidance or discussion would be greatly appreciated.

KevinCWLi, 2015-04-26:
Hey Phil

So right now I am accounting for this effect in my version of the sort code by simply storing a boolean for each ring/sector channel. When the sorting algorithm finds a valid event (the current condition only being that the energy of the front and back are within 10% of each other), the corresponding ring and sector channel which are involved in this "front-back" event are accounted for and are no longer used within the MMM sorting algorithm for that particular event. I therefore cannot double up on events… maybe you want to hardcode this?

padsley, 2015-04-26:
Yeah. That seems like probably the best way of doing it.

I was thinking about it and I realised that it might actually be possible to remove these events offline. All of the information required is present, right? I store the front and back energies, strip numbers etc. It should be possible to sort through the data offline (which I have to anyway to look at relative energy differences) and to take these things out. Then it's a problem iff the user wants to just draw spectra straight from the tree, but if they're interested in 8Be, they'll need to process offline and, more to the point, they will have to hard-code it in themselves.

OK. I shall work out (tomorrow) what the conditions which should be set in the offline tree are to take out 8Be/ghost hits and then I will try it with some actual data.

padsley, 2015-04-26:
Also, we both really have to find hobbies.

KevinCWLi, 2015-04-26:
I might be going off on a strange tangent here but bear with me:

With the method I described above, it is then possible for you to biase the matching ring and sector channels (when there are 2 ring signals and 2 sector signals that all validly match) depending upon how you sort through the channels. I intend, in the future, of using some simple algorithms I have seen online to randomly loop through the ring and sector channels per MMM……. 

This came up because it occurred to me that one side of the MMM is ever so slightly clover to the target in the PR226 setup. I was hoping to see it in the hit pattern - which I "did"….. however it happened to be the side of the MMM whose channel I begin looping over so this needs to be rectified….

KevinCWLi, 2015-04-26:
Hobbies can include SHARC cage diving.

padsley, 2015-04-26:
Boooo.

In the case of having two valid front and back hits (i.e., a ghost hit scenario), the two (I think) proper (I think) options are:

1. Match the front-back ADCs on the basis of the least difference between them. Whichever gives the smallest result out of the differences between front and back and hope that that's the right things.

2. Randomly match one of the front hits with one of the back ones when you realise that you have a ghost hit scenario. So, you find out that you have these two hits and then you say "OK... I have a 50% chance of keeping the original front strip and 50% chance of changing." Do the same for the back strip.

I personally favour the former but that's just my predilection.

padsley, 2015-04-26:
[image]

Also: Official BAGEL photo?

KevinCWLi, 2015-04-26:
Assuming that the "ghost hit scenario" comes from two correlated particles as in the case of 8Be, I think that option 1 is definitely better…. hopefully the silicon energy resolution allows for a sensible matching… if not then one basically randomising them anyways… this "effective randomisation" may be at the mercy of the accuracy of the front-back silicon calibrations as well right…

Did you take that picture yourself? Looks professionally done.
Actually that mouth looks like it is a naturally occurring pocket in the bread… you could argue for seeing the face of a saint? :)

padsley, 2015-04-26:
Off the top of my head, Philip is the Patron of pastries. Not quite bagels.

I didn't take it myself. I don't eat bagels. Only CAKE.