padsley/k600analyser #46: Crash PR226, run01108

padsley, 2015-04-27:
Currently have a crash when running this.

ptdc[1] seems to not be defined and this causes a segmentation fault. This has developed over the course of the day.

 nr of TDC datawords:                    0 words
L2472 (my line printout - unimportant)

 *** Break *** segmentation violation

===========================================================
There was a crash.
This is the entire stack trace of all threads:
===========================================================

Thread 2 (Thread 0x7f80ac142700 (LWP 27690)):
#0  0x0000003cbfa0ea5d in accept () from /lib64/libpthread.so.0
#1  0x00007f80b1c8e944 in TUnixSystem::AcceptConnection(int) () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libCore.so
#2  0x00007f80b08abd6c in TServerSocket::Accept(unsigned char) () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libNet.so
#3  0x00000000004166ce in root_socket_server (arg=<value optimized out>) at src/mana.cxx:5272
#4  0x00007f80adfd38fd in TThread::Function(void*) () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libThread.so
#5  0x0000003cbfa079d1 in start_thread () from /lib64/libpthread.so.0
#6  0x0000003cbf2e88fd in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7f80adfa91c0 (LWP 27683)):
#0  0x0000003cbf2ac65d in waitpid () from /lib64/libc.so.6
#1  0x0000003cbf23e609 in do_system () from /lib64/libc.so.6
#2  0x0000003cbf23e940 in system () from /lib64/libc.so.6
#3  0x00007f80b1c95298 in TUnixSystem::StackTrace() () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libCore.so
#4  0x00007f80b1c94113 in TUnixSystem::DispatchSignals(ESignals) () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libCore.so
#5  <signal handler called>
#6  0x0000000000431be9 in focal_event (pheader=0x7f80a1f38010, pevent=0x7f80a1f38020) at f-plane.c:2473
#7  0x000000000041caab in process_event (par=0x6913c0, pevent=0x7f80a1f38010) at src/mana.cxx:3155
#8  0x000000000041d042 in analyze_run (run_number=1108, input_file_name=0x7fff288eaa30 "/experiment-data/PR226/Data_B/run01108.mid.gz", output_file_name=<value optimized out>) at src/mana.cxx:4175
#9  0x000000000041d572 in loop_runs_offline () at src/mana.cxx:4363
#10 0x000000000041dda5 in main (argc=<value optimized out>, argv=<value optimized out>) at src/mana.cxx:5576
===========================================================

The lines below might hint at the cause of the crash.
If they do not help you then please submit a bug report at
http://root.cern.ch/bugs. Please post the ENTIRE stack trace
from above as an attachment in addition to anything else
that might help us fixing this issue.
===========================================================
#6  0x0000000000431be9 in focal_event (pheader=0x7f80a1f38010, pevent=0x7f80a1f38020) at f-plane.c:2473
#7  0x000000000041caab in process_event (par=0x6913c0, pevent=0x7f80a1f38010) at src/mana.cxx:3155
#8  0x000000000041d042 in analyze_run (run_number=1108, input_file_name=0x7fff288eaa30 "/experiment-data/PR226/Data_B/run01108.mid.gz", output_file_name=<value optimized out>) at src/mana.cxx:4175
#9  0x000000000041d572 in loop_runs_offline () at src/mana.cxx:4363
#10 0x000000000041dda5 in main (argc=<value optimized out>, argv=<value optimized out>) at src/mana.cxx:5576
===========================================================

[Analyzer,ERROR] [midas.c:1914:,ERROR] cm_disconnect_experiment not called at end of program
Cannot open message log file /home/adsley/k600analyser/midas.log

Apparently, there are no TDC words for this event. Which is a little odd.

padsley, 2015-04-27:
[Analyzer,ERROR] [mana.cxx:3961:ma_read_event,ERROR] Buffer size too small
/experiment-data/PR226/Data_B/run01108.mid.gz:255374  events, 11.03s

------------------------------------------------------
|                                                    |
|  TDC Events: 251474 
|  QDC Events: 251474 
|  ADC Events: 251474 
|  Scaler Events 3897 
| 
|  Events with no TDC datawords :       1   
|  Events with too many TDC datawords : 0 
|  Events with no QDC datawords:        1 
|  Events with no ADC datawords:        1 
| 
|  TDC chip ERRORS:     0 
|  TDC triggers lost:   0 
|  TDC buffer overflow: 0 
|                                                    |
------------------------------------------------------
 So, if affects event 255374?

padsley, 2015-04-27:
Maybe event 251474? Though that event looks fine:

------------------------ Event# 251474 --------------------------------
Evid:0001- Mask:0000- Serial:247636- Time:0x53a067db- Dsize:872/0x368
#banks:3 - Bank list:-QDC0ADC0TDC0-

Bank:QDC0 Length: 136(I*1)/34(I*4)/34(Type) Type:Unsigned Integer*4
   1-> 0xfa002000 0xf8004887 0xf810407c 0xf8014a9c 0xf8114081 0xf8024b26 0xf8124067 0xf8034af1 
   9-> 0xf8134078 0xf804403d 0xf8144095 0xf8054029 0xf8154077 0xf8064034 0xf8164089 0xf8074024 
  17-> 0xf8174092 0xf808403d 0xf818406c 0xf809403b 0xf8194076 0xf80a4032 0xf81a4094 0xf80b4043 
  25-> 0xf81b4077 0xf80c4043 0xf81c4068 0xf80d4038 0xf81d4075 0xf80e4032 0xf81e4059 0xf80f4040 
  33-> 0xf81f4099 0xfc03c763 

Bank:ADC0 Length: 560(I*1)/140(I*4)/140(Type) Type:Unsigned Integer*4
   1-> 0xfd000000 0xfa002000 0xf8004080 0xf810407d 0xf8014092 0xf8114077 0xf8024081 0xf8124089 
   9-> 0xf803408a 0xf8134076 0xf8044080 0xf814407c 0xf805408a 0xf8154069 0xf80640fe 0xf8164082 
  17-> 0xf8074738 0xf817407a 0xf80840a8 0xf818408f 0xf809409a 0xf819408d 0xf80a409b 0xf81a408c 
  25-> 0xf80b409b 0xf81b409b 0xf80c4087 0xf81c408d 0xf80d4067 0xf81d408e 0xf80e409a 0xf81e4084 
  33-> 0xf80f4082 0xf81f4086 0xfc03c763 0xfd000001 0xfa002000 0xf8004064 0xf81040a1 0xf8014060 
  41-> 0xf8114095 0xf8024066 0xf8124070 0xf8034055 0xf8134080 0xf804406f 0xf8144087 0xf8054059 
  49-> 0xf815408b 0xf8064088 0xf8164090 0xf8074055 0xf817408c 0xf8084060 0xf8184095 0xf8094077 
  57-> 0xf819409e 0xf80a407d 0xf81a4087 0xf80b407a 0xf81b409c 0xf80c4095 0xf81c4088 0xf80d4070 
  65-> 0xf81d4087 0xf80e4081 0xf81e409c 0xf80f408b 0xf81f407e 0xfc03c763 0xfd000002 0xfa002000 
  73-> 0xf8004074 0xf8104565 0xf801407e 0xf81140a8 0xf8024077 0xf8124045 0xf8034088 0xf8134056 
  81-> 0xf8044081 0xf814408c 0xf8054078 0xf8154052 0xf806408e 0xf816405f 0xf8074091 0xf81740b1 
  89-> 0xf8084091 0xf818404d 0xf8094078 0xf8194054 0xf80a4087 0xf81a407c 0xf80b4076 0xf81b4059 
  97-> 0xf80c40ac 0xf81c40d1 0xf80d407d 0xf81d405a 0xf80e408c 0xf81e404e 0xf80f406d 0xf81f4064 
 105-> 0xfc03c763 0xfd000003 0xfa002000 0xf8004050 0xf810406f 0xf8014050 0xf8114058 0xf8024065 
 113-> 0xf8124069 0xf8034066 0xf8134059 0xf8044054 0xf8144052 0xf8054041 0xf8154066 0xf8064046 
 121-> 0xf816406d 0xf8074049 0xf8174048 0xf8084053 0xf8184008 0xf8094041 0xf8194003 0xf80a4046 
 129-> 0xf81a409a 0xf80b4054 0xf81b4007 0xf80c4057 0xf81c4007 0xf80d405d 0xf81d4008 0xf80e4049 
 137-> 0xf81e4007 0xf80f404b 0xf81f4065 0xfc03c763 

Bank:TDC0 Length: 144(I*1)/36(I*4)/36(Type) Type:Unsigned Integer*4
   1-> 0xf8000000 0x4078ec7f 0x00132cb2 0x004b26fb 0x00034251 0x005326ec 0x005b2706 0x00632714 
   9-> 0x8000011f 0xf8000001 0x4078ec7f 0x00013592 0x8000007f 0xf8000002 0x4078ec7f 0x0001d438 
  17-> 0x8000007f 0xf8000003 0x4078ec7f 0x0000f4ac 0x8000007f 0xf8000004 0x4078ec7f 0x0006be82 
  25-> 0x8000007f 0xf8000005 0x4078ec7f 0x0003d855 0x8000007f 0xf8000006 0x4078ec7f 0x001561f0 
  33-> 0x007d7789 0x0005778a 0x01bd6474 0x800000df

padsley, 2015-04-27:
I'm not convinced that this isn't a problem with the reading in of data from /experiment-data. I can only access one file (the others are all non-readable due to permissions) and the code now suddenly quits before it should do.

This is non-ideal.

I might just jack this in and go home. I think that I have a backup of the data on my external hard-drive so I can try running from there.

padsley, 2015-04-27:
Ah. I think that the file is having a nice polite exit with:

[Analyzer,ERROR] [mana.cxx:3961:ma_read_event,ERROR] Buffer size too small

I really can't be bothered with this again at the moment.

padsley, 2015-04-27:
P.S. I just tried this with the master branch and it gives the same error (well, the segfault).

I have no idea what's happening... I'm giving up for now.

padsley, 2015-05-06:
I've just tried re-running this file (PR226/Data_B/run01108.mid.gz) through the code with the EIT guys having fixed the access for files on /experiment-data.

The code now runs cleanly through the whole thing.

==================================================================== 1108  

lut x1 offset: 0 
lut u1 offset: 0 
lut x2 offset: 0 
lut u2 offset: 0 
[Analyzer,INFO] Set run number 1108 in ODB
Cannot open message log file /home/adsley/k600analyser/midas.log
/experiment-data/PR226/Data_B/run01108.mid.gz:3620094  events, 460.47s

------------------------------------------------------
|                                                    |
|  TDC Events: 3565101 
|  QDC Events: 3565101 
|  ADC Events: 3565101 
|  Scaler Events 54991 
| 
|  Events with no TDC datawords :       0   
|  Events with too many TDC datawords : 0 
|  Events with no QDC datawords:        1 
|  Events with no ADC datawords:        0 
| 
|  TDC chip ERRORS:     0 
|  TDC triggers lost:   0 
|  TDC buffer overflow: 0 
|                                                    |
------------------------------------------------------
Cannot open message log file /home/adsley/k600analyser/midas.log

I'm putting this one down to the disk being a whiny little **** and I shall thus close the issue.

closed 2015-05-06 by padsley

KevinCWLi, 2015-05-06:
Ah good stuff!

I can't believe that that was the issue…. a real nightmare..

Kevin