padsley/k600analyser #47: Error analysing PR244 run 103

padsley, 2015-05-22:
This file has been run before on this machine without a problem. I.e. it's probably not a problem with the datafile but something else.

Cannot open message log file /scratch/data/SiDetectorTests/midas.log
                                                        ERROR: double ref times: module= 0, ntdc= 60, channel= 0
                                                        ERROR: double ref times: module= 0, ntdc= 60, channel= 0
bad tdc module nr 15
[Analyzer,ERROR] [midas.c:4609:bm_validate_client_index,ERROR] My client index 0 in buffer ']>,�?D$>4�@$<�B_' is invalid: client name '�p�', pid 1115460063 should be my pid 23121
[Analyzer,ERROR] [midas.c:4612:bm_validate_client_index,ERROR] Maybe this client was removed by a timeout. Cannot continue, aborting...
Aborted

After this error, attempted to analyse another file gives:

[Analyzer,ERROR] [midas.c:4960:bm_open_buffer,ERROR] Requested buffer size (100000) differs from existing size (390228)
Cannot open message log file /scratch/data/SiDetectorTests/midas.log

padsley, 2015-05-22:
Trying to run 'odbedit' to fix this:

[ODBEdit,ERROR] [midas.c:4960:bm_open_buffer,ERROR] Requested buffer size (100000) differs from existing size (390228)
[local:Default:S]/>Cannot open message log file /scratch/data/SiDetectorTests/midas.log
[ODBEdit,ERROR] [midas.c:4960:bm_open_buffer,ERROR] Requested buffer size (100000) differs from existing size (390228)
Cannot open message log file /scratch/data/SiDetectorTests/midas.log
odbedit: src/system.c:768: ss_shm_open: Assertion `(*handle) >= 0' failed.
Aborted

Kill me now. I'm going to go and do some cabling.

padsley, 2015-05-22:
FWIW: I just analysed run 102 without a problem. I had to clear out all of the ODB files (.SHM_* etc.) from the k600analyser directory and reload the ODB.

Going to try 103 again.

padsley, 2015-05-22:
[padsley@j150cswartz k600analyser]$ ./analyzer -i /experiment-data/PR244/run00103.mid.gz 
Root server listening on port 9090...

 ParameterInit

 CalibrationParametersInit
VDC1 is a new-type wire chamber
VDC2 is a new-type wire chamber
Using 2 terms for the ThSCAT position correction
Parameter number: 0	Parameter value: 1.000000e+00
Parameter number: 1	Parameter value: -2.322880e+00
Using 4 terms for the Y1 position correction
Parameter number: 0	Parameter value: 1.000000e+00
Parameter number: 1	Parameter value: -1.537930e-01
Parameter number: 2	Parameter value: 1.207840e-02
Parameter number: 3	Parameter value: -2.254860e-04
Using 3 parameters for the correction X position -> Rigidity calibration
Parameter number: 0	Parameter value: 3.776900e+00
Parameter number: 1	Parameter value: 3.577150e-04
Parameter number: 2	Parameter value: -6.002800e-10
Beam Energy: 196.200000 MeV
Going to do excitation energy calculation assuming inelastic scattering
Scattering Angle: 4.000000 degrees
mass1: 3728.400952 MeV/c**2
mass2: 22341.922650 MeV/c**2
ConfigEnd
Finished ReadConfiguration

PulseLimitsInit
ADCInit
QDCInit
ADCModules: 0
ADCsize: 0
TDCModules: 7
TDCsize: 896
Finished initialising parameters - to the sorting!
setupchannel2wireXUXU()
Running analyzer offline. Stop with "!"
Cannot open message log file /scratch/data/SiDetectorTests/midas.log

==================================================================== 103  

lut x1 offset: 0 
lut u1 offset: 0 
lut x2 offset: 0 
lut u2 offset: 0 
[Analyzer,INFO] Set run number 103 in ODB
Cannot open message log file /scratch/data/SiDetectorTests/midas.log
[Analyzer,ERROR] [mana.cxx:3961:ma_read_event,ERROR] Buffer size too small
/experiment-data/PR244/run00103.mid.gz:1287188  events, 102.63s

------------------------------------------------------
|                                                    |
|  TDC Events: 1284071 
|  QDC Events: 1284071 
|  ADC Events: 1284071 
|  Scaler Events 3116 
| 
|  Events with no TDC datawords :       0   
|  Events with too many TDC datawords : 0 
|  Events with no QDC datawords:        0 
|  Events with no ADC datawords:        1284071 
| 
|  TDC chip ERRORS:     0 
|  TDC triggers lost:   0 
|  TDC buffer overflow: 0 
|                                                    |
------------------------------------------------------
Cannot open message log file /scratch/data/SiDetectorTests/midas.log
Cannot open message log file /scratch/data/SiDetectorTests/midas.log

padsley, 2015-05-22:
[Analyzer,ERROR] [mana.cxx:3961:ma_read_event,ERROR] Buffer size too small <- This... isn't right.

KevinCWLi, 2015-05-22:
Voorspoed! (Good luck)

KevinCWLi, 2015-05-22:
Is this on your desktop Phil? Could it be that strange computer dependence again?

padsley, 2015-05-22:
==================================================================== 102  

lut x1 offset: 0 
lut u1 offset: 0 
lut x2 offset: 0 
lut u2 offset: 0 
[Analyzer,INFO] Set run number 102 in ODB
Cannot open message log file /scratch/data/SiDetectorTests/midas.log
                                                        ERROR: double ref times: module= 12, ntdc= 68, channel= 0
                                                        ERROR: double ref times: module= 12, ntdc= 68, channel= 0
                                                        ERROR: double ref times: module= 12, ntdc= 68, channel= 0
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
bad tdc module nr 12
[Analyzer,ERROR] [mana.cxx:3961:ma_read_event,ERROR] Buffer size too small
/experiment-data/PR244/run00102.mid.gz:298508  events, 24.00s

------------------------------------------------------
|                                                    |
|  TDC Events: 297825 
|  QDC Events: 297825 
|  ADC Events: 297825 
|  Scaler Events 681 
| 
|  Events with no TDC datawords :       0   
|  Events with too many TDC datawords : 0 
|  Events with no QDC datawords:        0 
|  Events with no ADC datawords:        297825 
| 
|  TDC chip ERRORS:     0 
|  TDC triggers lost:   0 
|  TDC buffer overflow: 0 
|                                                    |
------------------------------------------------------
Cannot open message log file /scratch/data/SiDetectorTests/midas.log
Cannot open message log file /scratch/data/SiDetectorTests/midas.log

padsley, 2015-05-22:
Correction: I did 102 but that had errors as well.

This is on my desktop. I'm going to let LCP know so he can have a look at the disk to check that nothing weird is happening.

padsley, 2015-05-22:
For completeness: 103 should have 1561893 entries and currently exits having run 1284071 entries.

padsley, 2015-05-22:
If I copy the file and try to run it locally, I get the same error for 103.

[Analyzer,INFO] Set run number 103 in ODB
Cannot open message log file /scratch/data/SiDetectorTests/midas.log
[Analyzer,ERROR] [mana.cxx:3961:ma_read_event,ERROR] Buffer size too small
run00103.mid.gz:1287188  events, 101.64s

KevinCWLi, 2015-05-23:
Awkwardly simply suggestion, but reinstall midas and then reset the odb? Because the fact that you can't run odbedit is quite crazy right….

Or maybe running this on your laptop if you haven't already.

gauss345, 2015-05-23:
Can you try to logout out completely, out of your desktop, and back in again. The permissions look ok , but I cannot use your krb login, so I cannot really test it. There is enough space on your desktop.
[root@j150cswartz SiDetectorTests]# df -h
Filesystem            Size  Used Avail Use% Mounted on
/dev/mapper/myvg-rootvol
                      458G   77G  357G  18% /

[root@j150cswartz SiDetectorTests]# ls -ltr
total 189512
-r--r--r-- 1 679 680 35968105 Nov 13  2014 run00281.mid.gz
-r--r--r-- 1 679 680 57208111 Nov 13  2014 run00282.mid.gz
-r--r--r-- 1 679 680 33242878 Nov 13  2014 run00283.mid.gz
-r--r--r-- 1 679 680  5438077 Nov 13  2014 run00284.mid.gz
-r--r--r-- 1 679 680 29887269 Nov 13  2014 run00285.mid.gz
-r--r--r-- 1 679 680 22015782 Nov 13  2014 run00286.mid.gz
-r--r--r-- 1 679 680 10264411 Nov 13  2014 run00287.mid.gz
-rw-r--r-- 1 679 680    22922 Jan 12 15:28 midas.log

the uid numbers is attached to your krb login. try to logout and back in again. the krb ticket only lasts 24 hours, then it expires, and you have to re-login or if you do not want to logout,

 do a 
command shell: kinit username; aklog;

to check the ticket time do:
commad shell: klist

padsley, 2015-05-23:
OK. I shall do it on Monday. I can't remotely access my machine from home.

padsley, 2015-05-25:
==================================================================== 103  

lut x1 offset: 0 
lut u1 offset: 0 
lut x2 offset: 0 
lut u2 offset: 0 
[Analyzer,INFO] Set run number 103 in ODB
Cannot open message log file /scratch/data/SiDetectorTests/midas.log
/experiment-data/PR244/run00103.mid.gz:1565614  events, 131.51s

------------------------------------------------------
|                                                    |
|  TDC Events: 1561893 
|  QDC Events: 1561893 
|  ADC Events: 1561893 
|  Scaler Events 3719 
| 
|  Events with no TDC datawords :       0   
|  Events with too many TDC datawords : 0 
|  Events with no QDC datawords:        0 
|  Events with no ADC datawords:        1561893 
| 
|  TDC chip ERRORS:     0 
|  TDC triggers lost:   0 
|  TDC buffer overflow: 0 
|                                                    |
------------------------------------------------------
Cannot open message log file /scratch/data/SiDetectorTests/midas.log
               
*opens bottle of gin and starts weeping*

neveling, 2015-05-28:
other than the gin consumption, did anything change with run 103?

tekuani, 2015-06-05:
A similar error has occurred on another machine using a transferred copy of run 103.

We should check the transferred run and the original to ensure that they're the same.

==================================================================== 103  

lut x1 offset: 0 
lut u1 offset: 0 
lut x2 offset: 0 
lut u2 offset: 0 
[Analyzer,INFO] Set run number 103 in ODB
Cannot open message log file /scratch/data/SiDetectorTests/midas.log
                                                        ERROR: double ref times: module= 0, ntdc= 63, channel= 0
                                                        ERROR: double ref times: module= 15, ntdc= 63, channel= 0
                                                        ERROR: double ref times: module= 15, ntdc= 63, channel= 0
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15

 *** Break *** segmentation violation

tekuani, 2015-06-05:
Error occurred again using "clean transfer file"?....

padsley, 2015-06-05:
I'm seeing the same thing at the moment:

Cannot open message log file /scratch/data/SiDetectorTests/midas.log
                                                        ERROR: double ref times: module= 0, ntdc= 63, channel= 0
                                                        ERROR: double ref times: module= 15, ntdc= 63, channel= 0
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15
bad tdc module nr 15

 *** Break *** segmentation violation

===========================================================
There was a crash.
This is the entire stack trace of all threads:
===========================================================

Thread 2 (Thread 0x7f2cea77f700 (LWP 9805)):
#0  0x0000003cbfa0ea5d in accept () from /lib64/libpthread.so.0
#1  0x00007f2cf02cb944 in TUnixSystem::AcceptConnection(int) () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libCore.so
#2  0x00007f2ceeee8d6c in TServerSocket::Accept(unsigned char) () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libNet.so
#3  0x00000000004163ee in root_socket_server (arg=<value optimized out>) at src/mana.cxx:5272
#4  0x00007f2cec6108fd in TThread::Function(void*) () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libThread.so
#5  0x0000003cbfa079d1 in start_thread () from /lib64/libpthread.so.0
#6  0x0000003cbf2e88fd in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7f2cec5e61c0 (LWP 9798)):
#0  0x0000003cbf2ac65d in waitpid () from /lib64/libc.so.6
#1  0x0000003cbf23e609 in do_system () from /lib64/libc.so.6
#2  0x0000003cbf23e940 in system () from /lib64/libc.so.6
#3  0x00007f2cf02d2298 in TUnixSystem::StackTrace() () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libCore.so
#4  0x00007f2cf02d1113 in TUnixSystem::DispatchSignals(ESignals) () from /afs/tlabs.ac.za/user/p/padsley/root/lib/libCore.so
#5  <signal handler called>
#6  0x0000000000431a23 in focal_event (pheader=0x7f2ce1620010, pevent=0x7f2ce1620020) at f-plane.c:2485
#7  0x000000000041c7cb in process_event (par=0x6912e0, pevent=0x7f2ce1620010) at src/mana.cxx:3155
#8  0x000000000041cd62 in analyze_run (run_number=103, input_file_name=0x7fff1301c370 "/experiment-data/PR244/run00103.mid.gz", output_file_name=<value optimized out>) at src/mana.cxx:4175
#9  0x000000000041d292 in loop_runs_offline () at src/mana.cxx:4363
#10 0x000000000041dac5 in main (argc=<value optimized out>, argv=<value optimized out>) at src/mana.cxx:5576
===========================================================

The lines below might hint at the cause of the crash.
If they do not help you then please submit a bug report at
http://root.cern.ch/bugs. Please post the ENTIRE stack trace
from above as an attachment in addition to anything else
that might help us fixing this issue.
===========================================================
#6  0x0000000000431a23 in focal_event (pheader=0x7f2ce1620010, pevent=0x7f2ce1620020) at f-plane.c:2485
#7  0x000000000041c7cb in process_event (par=0x6912e0, pevent=0x7f2ce1620010) at src/mana.cxx:3155
#8  0x000000000041cd62 in analyze_run (run_number=103, input_file_name=0x7fff1301c370 "/experiment-data/PR244/run00103.mid.gz", output_file_name=<value optimized out>) at src/mana.cxx:4175
#9  0x000000000041d292 in loop_runs_offline () at src/mana.cxx:4363
#10 0x000000000041dac5 in main (argc=<value optimized out>, argv=<value optimized out>) at src/mana.cxx:5576
===========================================================

[Analyzer,ERROR] [midas.c:1914:,ERROR] cm_disconnect_experiment not called at end of program
Cannot open message log file /scratch/data/SiDetectorTests/midas.log

padsley, 2015-06-05:
For reference: the version of the file transferred to @tekuani was through my machine so the same problem should probably occur in both cases assuming that the file isn't being read off the data disk correctly.

I think the solution last time was to restart my computer. :/

I'm going to try re-mounting /experiment-data for the moment.

padsley, 2015-06-05:
[user@machine k600analyser]$ sudo mount -a
[sudo] password for user: 
mount.nfs: access denied by server while mounting elbow:/experiment-data

Welp, fuck.

gauss345, 2015-07-06:
Hi padsley , could you tell me which odb settings you use and for which experiment. I've copied over your analyser code from your desktop, and will try to recreate the issue on my desktop. maybe there is an issue with 'gzread' function reading the files from nfs share. I will try tomorrow. I will try analise it with run00103.mid.gz.

gauss345, 2015-07-07:
ok...

latest midas stable version. (git checkout)
root v5.34.32 
gcc version 4.9.2 (Debian 4.9.2-10) 

 mount netapp2:/vol/experimental_data/experiment_data /experiment-data -o ro,wsize=32768,rsize=32768

mounting the same directory from netapp2 , onto my desktop, yields the results
below.

is this correct ?

./analyzer -i /experiment-data/PR244/run00103.mid.gz 
.
.
.
/experiment-data/PR244/run00103.mid.gz:1565614  events, 96.71s

------------------------------------------------------
|                                                    |
|  TDC Events: 1561893 
|  QDC Events: 1561893 
|  ADC Events: 1561893 
|  Scaler Events 3719 
| 
|  Events with no TDC datawords :       0   
|  Events with too many TDC datawords : 0 
|  Events with no QDC datawords:        0 
|  Events with no ADC datawords:        1561893 
| 
|  TDC chip ERRORS:     0 
|  TDC triggers lost:   0 
|  TDC buffer overflow: 0 
|                                                    |
------------------------------------------------------

padsley, 2015-07-07:
Those results look correct. One of the problems has been that the problem is intermittent. I've been able to sometimes use the datafiles correctly and sometimes not. Depends on which way the wind is blowing.

I realise also that I gave no useful information when I posted the thing on redmine which was a stupid oversight on my part. I will try to collect what information I have and post it here tomorrow. I'm off work again today and am taking a break from Bring(ing) Up The Bodies by looking at Github. Obviously.

gauss345, 2015-07-08:
just by-the-by... I realized reading the k600analyser, that there isn't much of a coding convention followed, and makes for very difficult reading. It might help to at least get the 'official' versions of software used by the group, into a style everyone can read. Also, maybe link/give info, on the daq.tlabs.ac.za site?

padsley, 2015-07-08:
Yeah. This is in the on-going list of tasks that I haven't managed to do.

I have tried (and possibly failed) to make the silicon stuff fairly consistent (and the Parameters.c section of the code as well). However, there is not really any documentation yet. Ooops.

f-plane.c has had a large number of changes and additions over time and could do with being tidied up. However, this is non-trivial as it's somewhat complex. I can try having a look at making things their more consistent at some point but I also have to analyse the data from the previous experiment so spending two weeks on rationalising things will have to wait for quite a while.

gauss345, 2015-07-08:
well I'm doing it anyways for other things. It would be good for the conventions in all the software for k600, at least the official ones, follow the same style and conventions. I did rewrite the analyser a while back, into c++.

gauss345, 2015-07-13:
I'm starting to look at config.cfg and putting it into sql db form. Suggestions would be nice. and I guess this isn't part of issue #48 anymore, so should probably be in another thread.

padsley, 2016-02-15:
I'm going to close this issue as well.

I think that the problem may have been machine/disk-reading dependent and there's no point leaving this as a warning in case something goes wrong later on.

I think that we should leave this for now.

closed 2016-02-15 by padsley