padsley/k600analyser #54: W1 sort efficiency

padsley, 2015-07-07:
I need to change the W1 sort to be the same style as the MMM sort so that it's quicker. This needs to be done before Christian's experiment as he wants 6(?) W1 detectors which is 192 channels and the computing cost of this sort scales as N**2.

I'll do it myself but help would be appreciated.

nichubbard, 2015-07-14:
Preliminary update: I have added the MMM style conditions to the W1 sort in setupdev and uncommented the define in f-place.c (is this meant to be required?). Most of the sort time seems to be spent on the focal plane stuff.

Currently the new conditions seem to be throwing away far too much of my data, so am investigating what conditions seem to be failing

padsley, 2015-07-14:
Yup. If you don't uncomment the #define in f-plane, it will just ignore all of the W1-type data.

You might need to check the channel mapping in the input configuration file. It might well be wrong.

nichubbard, 2015-07-14:
Okay, the issue was a silly typo, everything seems quite reasonable now. I will push the changes to setupdev soon so people can use it/check it works

padsley, 2015-07-14:
Bosh. Thanks for that.

If you're happy with how it's working then close the issue. If not, then would you like us to do any tests?

nichubbard, 2015-07-14:
I will be double checking tomorrow that nothing seems to be causing problems, and then will close it. If you want to test on any data that'd be fine - I don't have any data with back TDC channels so those conditions haven't been tested if you happen to have any.

padsley, 2015-07-14:
I don't have any W1 data with that.

We can test it with data when you're down in August/September.

nichubbard, 2015-07-15:
Okay I have checked with more data, and it looks like it's working fine,

closed 2015-07-15 by nichubbard

nichubbard, 2015-07-21:
I have just pushed a quick bug fix to not loop over back TDC channels if they aren't being used, as otherwise you just get a lot of duplicated events