padsley/k600analyser #55: K600Analyser rewrite to ROOTANA.

gauss345, 2015-07-13:
Hi.

This is just to start a conversation regarding rewriting the analyzer in a more modular form (C++),
that is ROOT based.

gauss345, 2015-07-13:
For some who do not know what rootana is, can find it here: http://www.bytebucket.org/tmidas/rootana 

You do not require Midas installed to used it, but you do require ROOT. It is pretty nicely defined, with examples , and these days, it is very well "frameworked" .

KevinCWLi, 2015-07-13:
It shall prove a good learning experience for us students to observe this phase of the analyser development as good consistent coding practice is paramount (personally I need to make sure mine is up to scratch…).

I have not really heard of ROOTANA before… just visited the page now. Why is it particularly beneficial over the existing form of the analyser? I get that the "standalone" aspect is good, but is it really worth the trouble? (sorry if this is obvious, but it is not to me)

gauss345, 2015-07-13:
> Why is it particularly beneficial over the existing form of the analyzer? 

Currently, the k600analyser comes in various forms, for offline analyses we have this version[s] ( k600analyser ), and for online analysis there is the rome analyzer. With this approach, using ROOTANA, you get that both online and offline analysis are done via the same framework, which helps with code reuse-ability and redundancy.  But, one of the major differences is that you do not require ROOT ( not recommended ) or MIDAS to analyze. Having to maintain those 2 packages is an unnecessary overhead for students/users who just want to analyze data, and do not really care for the extra load that this brings. 

btw, my browser is moaning about the words analyser vs analyzer. ( americans ). :( 

>I get that the "standalone" 
> aspect is good, but is it really worth the trouble? (sorry if this is obvious, but it is not to me)

Well... it is a preference, that makes working with ROOT easier. Everything is in C++. ;) 

 http://trshare.triumf.ca/~gchristian/dragon/analyzer/html/index.html <--- see here for more info.

They've basically beaten us to the punch sadly, but it is mostly where k600 daq analyses is heading. Our setup will be different for various reasons, but, the ROOT integration will be similar.

In the end ( soon soon ), if we get stuff finished, you'll get a root tree structure containing the data, instead of a .mid (raw) file. That is easier to handle, and traverse, and lessons the code (written) to analise.  ;) 

I'm very sleepy, so I hope I make sense here... :( But feel free to correct. or make suggestions.