padsley/k600analyser #56: definitions of nr of QDC ADC TDC channels

neveling, 2015-07-14:
There are some definitions in analyzer.h  and also some in Parameters.c
I suggest we do away with those on analyzer.h and only use Parameters.c

padsley, 2015-07-14:
I suggest that this is left to Retief as he's the person who knows where things are. When he's finished with his other f-plane.c mods, we can pull that information over from his branch into the main body of the code.

neveling, 2015-07-15:
The main modifications to f-plane regarding the addition/modification of raytracing code is now to be found on the branch SetupDevRetief.

I also made a few changes to qdc.c and adc.c to make sure the definitions in analyzer.h is no longer needed and rather the numbers in Parameters.c are used.

In addition I started some general cleanup of f-plane.c.

While the cleanup will continue for some time I think Phil you can look into merging this branch with your SetupDev branch?

In any case, I think before a future major rewrite of the analysis code it would be good if the existing code is cleaned up a bit. It will make the rewrite so much easier.

closed 2015-07-15 by neveling