padsley/k600analyser #59: ODBedit Load Error

vpesudo, 2015-07-23:
While trying to use odbedit to load the odb file, the following error was returned:

[vpesudo@ k600analyser]$ odbedit
Cannot open message log file '/scratch/data/SiDetectorTests/midas.log', open() errno: 2 (No such file or directory)
[local:Default:S]/>load pr226-we2.odb
[ODBEdit,ERROR] [odb.c:6325:db_paste,ERROR] found unknown data type "" in ODB file
Cannot open message log file '/scratch/data/SiDetectorTests/midas.log', open() errno: 2 (No such file or directory)

vpesudo, 2015-07-23:
Though the analyser now runs... not sure what the 'unknown data type' error was doing.

padsley, 2015-10-13:
For reference:

The first error "Cannot open message log file '/scratch/data/SiDetectorTests/midas.log', open() errno: 2 (No such file or directory)" isn't a problem. It just means that the log file directory can't be found. That's because it's set to be on my machine for the silicon detector information.

The second one (unknown data type) I've seen before. Again, I don't think that it's usually a problem.

If it reoccurs, then delete the .odb file and reload a backup version.

closed 2015-10-13 by padsley