padsley/k600analyser #62: Mac OS Makefile

pellegri, 2015-08-18:
Need mods to this now: clang no longer supports -lrt compilation option.

Need to force the makefile to distinguish between linux and mac OSes at compile-time.

KevinCWLi, 2015-08-18:
Phil had a way of doing this already if I'm not mistaken. If memory serves me correctly, t is the same method that is used in the makefile of midas.

By the way, does this mean that you have gotten the analyser to work on your mac Luna?

padsley, 2015-08-18:
We're working on that at the moment.

I do have a way of doing this but I need to implement it for the makefile wrt lrt and clang.

gauss345, 2015-08-19:
I have a response from midas ppl... 
"dd1

This is MacOS 10.10 or 10.9? We build, run and debug MIDAS on both, except for the analyzer part. The mana.c analyzer is very out of date wrt modern programming styles and modern analysis tools. For new experiments we direct people to use the ROOTANA analyzer or the ROME analyzer, both linked from https://midas.triumf.ca. The problem you see in the mana.c based analyzer should be fixable, but you have to send us more information than "it crashed" (also, there is no multithreading in mana.c, maybe there is one thread created for the data server, if enabled). K.O.
"

padsley, 2015-08-19:
I'm not sure if this is the same problem.

One of these problems is clang rather than the analyser/ROOTANA etc.

It has occurred to me that this shouldn't be a Mac OS thing. It should be that the Makefile detects if you're compiling with clang, that it chooses the correct commandline options for compilation.

I think.

gauss345, 2015-08-20:
@padsley What has occurred with you ? The makefile is setup to correctly compile for Mac OS. clang and gcc use similar command line switches. The differences in the two compilers is not what is causing this issue, afaik. You would have to explain further what you mean.

closed 2016-07-27 by padsley