padsley/k600analyser #66: Suppress Data classes if no filling

pellegri, 2015-08-20:
Found a bug: need to suppress the clearing of data classes and deleting of the pointer when there has been no filling of the data structure.

e.g.,

#define _SILICONDATA
//#define _MMM
//#define _W1

This will declare a SiliconData pointer but nothing else. If you do this, then at the end of the code, when it tries to delete the SiliconData class, there will be a crash.

The code should prevent this by testing whether anything has been done to the pointer or by requiring that _SILICONDATA is coupled with at least one actual silicon data processing thingy.

gauss345, 2015-08-22:
" The code should prevent this by testing whether anything has been done to the pointer or by requiring that _SILICONDATA is coupled with at least one actual silicon data processing thingy. "

It is good practice to declare a null pointer when initializing the pointer. 

checking the pointer, ie. if( pointer ) then delete class/free(pointer) . I hope this helps.

padsley, 2015-08-22:
It does make sense. I tried doing that but with the opposite condition. Doh.

I'll make the fix next week. Thanks for that. The pointers are (if I remember correctly) declared null but I was failing to use the correct test conditon.

gauss345, 2015-08-23:
Below is what would be considered bad practice using preprocessors.

here you have 3 #define statements which are all coupled due to the "*si" pointer. 
It is also not clear from the #define statements the interdependence on each other.

Why does the pointer need to be global ?

I do not see any checks in  place re size of pointer etc?

in C/C++, it is assumed the programmer will take care of memory and pointer
management. 

also preprocessors shouldn't really be used in this fashion. 

:)  :+1: 
 

#define _SILICONDATA 
#define _MMM
//#define _W1

#ifdef _SILICONDATA
SiliconData *si;
#endif

#ifdef _MMM
    if(si)
    {
      si = MMMSiliconSort(ADC, TDCHits, TDC_channel_export, TDC_value_export);
    }
#endif

#ifdef _W1
    if(si)
    {
      si = W1SiliconSort(ADC, TDCHits, TDC_channel_export, TDC_value_export);
    }
#endif

#ifdef _SILICONDATA
   si->ClearEvent();//Clear the SiliconData gubbins at the end of the event in order to make sure that we don't fill the disk up with bollocks
   delete si;//Delete the pointer otherwise we lose access to the memory and start to crash the machine
#endif

padsley, 2015-08-24:
Next thing you'll be telling me that I shouldn't use the word 'bollocks' in the code...

padsley, 2015-08-24:
In answer to some of your questions:

I know that the various interdependences aren't clear. That's my fudging to get things working. I'm not quite sure how to tidy this up but I realise that it should be. I shall attempt to do so but will probably delay that task until Retief has finished the rationalisation of f-plane.c.

The pointer needs to be global unfortunately. I use it to define part of the output TTree which is declared from the SiliconData class. When that is done, you tell the code that the place that it should look for the SiliconData stuff is the pointer 'si'. If there is a way of doing this without a global declaration, that'd be grand but my (slightly wonky) understanding of ROOT leads me to think that this isn't the case.

For checking the size of the pointer: I'm not sure what you mean by that. Is the pointer always the same size as it's just a pointer or does it change with the 'contents' of the pointer? If it's the latter, then it won't work because the whole reason that I use the SiliconData class as is is that is has variable size stuff so that it's possible to store an arbitrary number of silicon hits within each event.

If someone doesn't come in with a working card for the tea room in the very near future, I'm going to end up eating a teabag.

gauss345, 2015-08-24:
"bollocks" is fine... my code is littered with the word `jebus` ;)

gauss345, 2015-08-24:
" I do not see any checks in place re size of pointer etc? " 

Sorry, missing words,  'size of what the pointer is pointing to' ? to make sure it is in fact pointer to SiliconData. 

It was a late night ... :(

padsley, 2017-02-01:
I'm assuming that this is done. If not, it's been sitting around for a while and no one is complaining now.

closed 2017-02-01 by padsley