padsley/k600analyser #67: Detector numbering

padsley, 2015-08-26:
Just a random thought: should we try to make detector numbering match the experimental numbering? Otherwise the W1 in detector position 3 has to be entered as detector 1.

It might make the config file simpler/easier to use at the cost of breaking something else somewhere else.

nichubbard, 2015-08-26:
Would this affect the DetectorHit value as well? I think it makes sense for that at least to match experimental numbering

padsley, 2015-08-26:
I think so, yes.

For the current case it's not too much of a problem until we make a separate X1 class and the W1 numbering has to start at 3.

I'll come up with a scheme for changing it and get around to it in the nearish future... unless anyone else would like to try.

padsley, 2015-10-13:
The answer has occurred to me:

There should be a separate detector number and detector index.

The index will start at 0 but the number doesn't have to. Then, when you do things in the sort code, you deal with the index (which will be where things sit in arrays) but the detector numbering in experimental not C.

This should be implemented in the SetupDev branch at some point.