padsley/k600analyser #70: To keep in mind if using more than 7 TDCs: JJAutotrim issues

neveling, 2015-10-02:
This is relevant if you want to create CableLength.dat via the JJ_AUTOTRIM preprocessor
mechanism

If you use more than 7 TDCs  you have to make the following modification
since hTDC_REF is defined as:

static TH1F *hTDC_REF[TDC_CHANNELS];   

and TDC_CHANNELS=896

Add
if(channel<895) as shown below

#ifdef _JJAUTOTRIM  
if(tof>gates.lowtof && tof<gates.hitof && PaddlePIDGatesFlag==1){		// if you want to use JJ_autotrim.C
      if(channel<895) hTDC_REF[channel]->Fill(ref_time);                     // fill spectra for individual wires   	
}											
#endif

closed 2015-10-02 by neveling

padsley, 2015-10-02:
2 things:

1. I've made this mistake already. :( JJ's script Does Not Like silicon data.

2. Is this line complete: 
{
if(tof>gates.lowtof && tof if(channel<895) hTDC_REF[channel]->Fill(ref_time); // fill spectra for individual wires 
}

??

neveling, 2015-10-12:
Long overdue answer:

      
#ifdef _JJAUTOTRIM  
if(tof>gates.lowtof && tof<gates.hitof && PaddlePIDGatesFlag==1){
         if(channel<895) hTDC_REF[channel]->Fill(ref_time);      
}											
#endif