padsley/k600analyser #71: Master branch

padsley, 2015-10-13:
@pellegri has noted that the master branch doesn't compile.

This problem has been noted before.

It's due to a mistake that I've made in the definition of PulseLimits.

I need to fix it at some point. This is a note to remind me to fix it.

padsley, 2015-10-21:
I can't reproduce this error on my laptop.

If someone else could try compiling the master branch to see if it produces an error and could then post the log here, I'd be grateful.

padsley, 2015-10-21:
I merged the fixed version of SetupDev that Alex forked back into the master as it went in there easily and probably is sufficient for what anyone might need.

This means that the master branch should be the version using .cfg files that I have been setting up on arthur at iThemba.

closed 2015-10-21 by padsley

neveling, 2015-10-22:
Sorry I am a bit late
Anyway,  I compiled it with not problems, no errors

R

padsley, 2015-10-22:
Grand.

I'm going to assume in lieu of any comments to the contrary that the master branch is a fully armed and operational battle station.