padsley/k600analyser #73: MMM Phi Calculation

padsley, 2015-10-18:
The calculation of the phi value in MMM.c is totally incorrect.

I'm honestly not entirely sure what I did based on how it looks. I must have been having some form of seizure.

I need to fix this fairly soon.

padsley, 2015-10-21:
For the moment, I'm just commenting out the error condition on this. I intend to put it back in at some point soon when I've fixed the code but I need to work out how to do that first. Also, for the PR242 setup, the phi values differ from those of the PR244 and PR226.

I need to consult @KevinCWLi about how best to do this and he's on a work trip at the moment. It shall be done upon his return.

padsley, 2015-10-21:
For the moment, I've suppressed the phi calculation as well (i.e. phi=0 always for the MMMs).

neveling, 2016-03-09:
The calculation of angles for the MMM detectors  can be done as a simple lookup-type action, i.e. depending on the detector, sector and ring number we take a value for the phi and theta from a 
list of values supplied from Kevin, as calculated in his simulations. I will implement this later today

KevinCWLi, 2016-03-09:
1.) Hey Retief

Funny coincidence, the calculation for the W1-array in PR228 has just been completed. It is still preliminary and I have attached it below.

2.) Hi Vicente, the total solidAngle is 2.36% of 4pi.

3.) Hi Nic

I have attached a header file which provides the geometrical information for the backward angle detectors for PR228.
For detector x, row y and column z (i.e. a particular pixel(x,y,z,)):

the average theta : GA_W1[x][y][z][0],
the average phi : GA_W1[x][y][z][1],
the total solidAngle : GA_W1[x][y][z][2].

Nic I have tried to make sure that the calculation matches the channel setup in PR228, i.e. theta varies predominantly with row and column with phi.

If you find a problem, just email me.

Kind regards
Kevin