padsley/k600analyser #75: W1FrontBackTest and Absolute Energies

nichubbard, 2015-10-26:
I have traced a problem with the sort for PR228b producing horrible SiliconTime spectra to using an absolute energy cut inside W1FrontBackTest, instead of the old relative one. By adding lots of printf statements and looking at the output, it seems that the absolute cut causes a lot of combinations of (front,back) to pass the test.

I have reverted back to the relative condition for now, as it seems to work better but it seems the choice of sigma for absolute cuts needs some tuning, or something.

nichubbard, 2015-10-26:
To be more specific, the tests are:

relative: diff/(0.5*(FrontEnergy+BackEnergy)) < 0.05
absolute: abs(diff) < 3*sigma

I had sigma set to 30 keV in my code it seems

padsley, 2015-10-29:
This doesn't make sense to me, really. Unless it's passing loads of stuff just at low energy.

I think that we need some proper diagnostics on it. I'll have a look next week at some point.

I would have thought that the proportional condition would also select a load of low-energy stuff.

nichubbard, 2015-10-30:
It might actually be the opposite, things look okay with the test completely skipped so it might be the absolute test is too narrow.

I'll try and look at some raw front-back data and see what it looks like

padsley, 2015-11-04:
I'm just updating things on the issues.

Is there an update on how this is going?

One thing to consider: impose a condition on the energies of the hits involved. It might be low-energy noise which is screwing things up. As the relative front vs back energy will become extremely narrow at low energies, a large amount of noise might be suppressed that isn't when the absolute test is used.

nichubbard, 2015-11-04:
I think that it was due to a too narrow cut on Front/Back when absolute. I raised sigma to 40 keV (so max. difference 120 keV) and things look a lot more reasonable. Looking at Front/Back graphs it seems the majority of low energy data has a difference of less than 100 keV or so, so the 90 keV cut before might have cut out a fair bit of the electron signal, but keeping more of the low energy noise, making the TDC spectra look bad.

padsley, 2015-11-04:
Schweet.

If you're happy with things, close the issue. Then I won't try to fix it at some point next week. :P

closed 2015-11-04 by nichubbard