padsley/k600analyser #76: Data classes with multiple detectors

padsley, 2015-11-02:
I'm an idiot.

padsley, 2015-11-02:
More detail: @pellegri says that I'm an idiot.

pellegri, 2015-11-02:
I've just find a way to point it out :P

padsley, 2015-11-02:
The problem is that the output class pointer (e.g. gammy) is just set to the last working version of the thing that has been filled. This wrong and evil and bad.

It should be set so that the same GammaData (for example) object is used throughout and different sorting algorithms just add to that pre-existing class.

This will require some (hopefully not too major) rewrite of the sorting algorithm.

In future, the pointer to the Gamma/SiliconData class should be handed to the sort algorithm (which should be a void not a pointer to a Gamma/SiliconData class) so that it can just add to the bottom of something that's already there rather than resetting it.

padsley, 2015-11-02:
Also, ow.

padsley, 2015-11-02:
This change will be made in the SetupGammaDev branch and then transported across into the master when possible.

padsley, 2015-11-04:
I've done this for the GammaData classes in the SetupGammaDev branch.

When @pellegri has finished her development work on that branch and we merge it back into the master, the change will make it to the master.

I will make the same change to the SiliconData classes in the master branch at some point fairly soon. It's not urgent as no one requires it (yet).

padsley, 2015-11-27:
This was done at the same time as the SetupGammaDev merge back into the master branch, and was expanded to include the silicons.

closed 2015-11-27 by padsley