padsley/k600analyser #84: W1.c Crash

padsley, 2015-12-02:
Trying to see how the W1 data look for PR231, we're seeing a crash.

Mismatched size: Theta
Mismatched size: Phi
Mismatched size: Time
Mismatched size: Front Time
Mismatched size: Back Time

 *** Break *** segmentation violation

This implies that something is going wrong when SiliconData is being constructed in W1.c. I think.

I'm trying to chase it down now. Nic: W1.c has been modified by you. Can you think of any reason why only some of the data entries for the class would be being filled?

nichubbard, 2015-12-02:
I can't think of a reason off the top of my head, and just looking at the W1.c code, all the SiliconData fills occur together with no ifs around it, so they should all be synchronised, and those 5 entries are definitely filled, so not sure why they are being mismatched. Could it be related to the recent change to not create SiliconData inside the W1 sort function?

neveling, 2015-12-04:
The way that the si pointer is handled seemed to cause the problem.

I noticed that for events that did not have a good W1 event (in this case an event with valid TDC channels but not associated valid ADC channels) a segmentation fault would result when the code attempts to delete the pointer, which happens in the last few lines in f-plane.c
The same problem does not happen for an event with good W1 data

In the current version of the code on the master branch the assignment of the si pointer is
as follows:

W1SiliconSort(ADC, TDCHits, TDC_channel_export, TDC_value_export, si);

whereas it used to be:

si=W1SiliconSort(ADC, TDCHits, TDC_channel_export, TDC_value_export);

By changing back to the previous version I could get the code to run without a segmentation error.

padsley, 2016-05-12:
I think that is the same problem as JJ has been having and it's something that is now fixed. Closing issue.

closed 2016-05-12 by padsley