padsley/k600analyser #86: Calculation of ThFP, ThFPx and ThSCAT

padsley, 2016-02-07:
When we add the proper sections to the code to force the reading of the VDC configuration, we should also make sure that ThSCAT is being calculated using the correct formalism (i.e. using ThFPx if an old VDC was used).

This was a problem for @along4's analysis and will probably have emerged in other places as well.

padsley, 2016-02-08:
Thinking about it, I think that the code already includes the structure for automation of the VDCs... it's just not yet implemented.

I'm leaving this task until the new f-plane is written as I'm not sure what might change.

neveling, 2016-06-01:
Some work on this issue was done recently. Will follow up and summarize soon

closed 2016-07-27 by padsley