padsley/k600analyser #90: SiliconHits is too high - crashing

padsley, 2016-03-07:
@neveling noticed a problem with run 1009 from PR226 - the second event crashes the analyser.

The problem seems to be manifested in the silicon event size being: 229376

Which is wrong.

Currently investigating.

padsley, 2016-03-07:
Huh.

I don't have any problem, using the version of the code I got from the repo.

padsley, 2016-03-07:
Argh. I hate computers. Lovely lovely beer. Beeer.

neveling, 2016-03-07:
It happened because I had this line almost at the end of main.c uncommented for some reason:
//delete si;    //Delete the pointer otherwise we lose access to the memory and start to crash the machine

Should this cleanup not happen inside the EOR routine?

closed 2016-03-07 by neveling

padsley, 2016-03-07:
Oh! That's my fault.

Yeah... so, if we delete the pointer, the code flakes out.

I think that, yes, it should happen at the end of the run, especially as the pointer is cleared in at the end of each event. In any case, then the program finishes, then the memory will be freed (I think?) so that doesn't matter too much.

I have a feeling that I needed to delete it when I was structuring the plugins in a different manner. Now that I've changed it (from returning pointers to being voids, like my social life) then they shouldn't be deleted.