padsley/k600analyser #91: Phi and Phi: a comment on coordinate systems

neveling, 2016-03-09:
In the latest version of the code there are two phi angles: one (the original one) is a phi in a cartesian coordinate system, i.e. the out-of-plane angle.
I suggest we change the name of the old one to something that will clearly indicate vertical scattering angle, or at least which coordinate system it is we are using.
And, I take it thetaprime and phiprime refer to the scattering angles, not the focalplane angles?
And by the way, the thetescat values (from the original theta calculation) refers to the angle relative to Theta-K600.

padsley, 2016-03-09:
Mea culpa.

If we can have something consistent, then that's probably good.

ThetaSCAT refers to scattering relative to the K600 aperture. Maybe phiSCAT can be the same: the y-component into the aperture?

Then phi is the actual *real* phi value... and theta the true theta.

I think that my thetaprime values were, yes, to the scattering angles not the focal plane angles. I should probably clean that up to remove any possible confusion.

neveling, 2016-03-09:
Ok, then we have
thetaFP and phiFP
thetaSCAT and phiSCAT  (currently thetaprime and phiprime)
theta and phi
This also brings us to another thing that needs standardization: names of leaves, specifically with relation to the use if capitol letters:
thetaFP, or thetaFp, or thetafp, or ThetaFp etc etc etc
For now I go with the 6 names in the beginning of this paragraph: i.e. the thetaFP version of it...

padsley, 2016-05-12:
I don't think that this is still an open issue... rather it's something useful for future reference. I'm going to close it, therefore.

closed 2016-05-12 by padsley