padsley/k600analyser #95: Deleting pointers

padsley, 2016-05-10:
Is this fixed for the new master and for gamma data? Asking for a friend.

Has been a problem with SiliconData. Should make sure not a problem for GammaData.

neveling, 2016-05-12:
I believe the Si pointer is no longer being deleted. This is how it looks like now in main.c
The pointer for the gamma data is still being deleted though. Not a problem for me so far as I do not use that in my recent analysis work

#ifdef _SILICONDATA
si->ClearEvent(); //Clear the SiliconData gubbins at the end of the event in order to make sure that we don't fill the disk up with bollocks
#endif
#ifdef _GAMMADATA
gammy->ClearEvent();//See comment above about GammaData::ClearEvent()
delete gammy;//See comment above about deleting *gammy
#endif

jjvz, 2016-05-24:
Note on W1.c:
I had to add the lines (similar to what is in MMM.c): 

extern double *TDCOffsets;

and

si->SetOffsetTime(mTDC.GetValue(k) - TDCOffsets[mTDC.GetChannel(k)]);

... so that the TDC offsets mismatch error does not pop up endlessly when analysing a run.

padsley, 2016-05-25:
Oops.

I have a feeling that I had implemented that for MMM.c but not W1.c.

I also need to make sure that the TDCOffsets default to 0 when not defined and I've not done that yet. I'll open a new issue and deal with that later.

jjvz, 2016-05-25:
No problem... also a bunch of stuff in W1.*, HagarData.* and GammaData.* that I am commenting out that are not relevant to Hagar, that are e.g. giving empty histograms (...probably the reason I'm getting a flood of error messages (Error in <TH2F::Fill>: Invalid signature - do nothing), or are crashing the analyser because of inconsistent declarations.  
I will post these changes soon... as soon as I have picked them all up.

padsley, 2016-05-25:
OK.

In that case, can you make a new issue each for the W1, Hagar and GammaData
errors? Some of them are (possibly) Luna's responsibility and some of them
are my responsibility.

Dr Philip Adsley MA MSci (Cantab)
Postdoctoral Research Fellow, University of Stellenbosch/iThemba LABS

jjvz, 2016-05-28:
In main.c, comment out "delete gammy;" i.e.:
#ifdef _GAMMADATA
   gammy->ClearEvent();//See comment above about GammaData::ClearEvent()
//   delete gammy;//See comment above about deleting *gammy
#endif

Otherwise my analyser crashes. I put the "delete gammy" line in the "main_eor" routine lower down, and it works fine, i.e.:
INT main_eor(INT run_number)
{
#ifdef _SILICONDATA
   delete si;        //Delete the pointer otherwise we lose access to the memory and start to crash.. 
#endif
#ifdef _GAMMADATA
   delete gammy; 
#endif

closed 2017-03-03 by padsley