padsley/k600analyser #100: Error in the master branch while running PR254 calibration data

pellegri, 2016-05-27:
This is the error aiutoooooooo:
Error in <TH2F::Fill>: Invalid signature - do nothing

jjvz, 2016-05-28:
We found the 2D histogram complaining was hTDC2DModule[NR_OF_TDCS]. In main.c it is defined as an array of NR_OF_TDCS elements, which is hard set to 7 in main.c, line 243 (?): 
const int NR_OF_TDCS=7;

Now I use 8 TDC modules, as I have set correctly in my config.cfg file, but we cannot define the array with a external varable (wel I cannot...) such as: NR_OF_TDCS=TDCModules, this does not work. So i changed NR_OF_TDCS=8; and now it does not print the TH2F warnings anymore.... Retief?

padsley, 2016-05-29:
The fix for this is something akin to da followin:

 TGraphErrors *gEnergy[npeaks];
  TGraphErrors *gDCS[npeaks];
  for(unsigned int i=0;i<npeaks;i++)
  {
    char buffer[256];
    sprintf(buffer,"gEnergy_%d",i+1);
    gEnergy[i] = new TGraphErrors();
    gEnergy[i]->SetName(buffer);
    gDCS[i] = new TGraphErrors();
    sprintf(buffer,"gDCS_%d",i+1);
    gDCS[i]->SetName(buffer);
  }

...but... y'know... for TH2Fs and stuff.

neveling, 2016-05-30:
Mmm, maybe I am just stoopid (no comments please)  but I cannot see how this will work.
Currently in main.c the histograms are declared between lines 252-330, giving them scope across all routines.
Definition/initialization of these histograms happens inside the initialization routine main_init, around line nrs 500-ish to 600-ish.

Currently the declaration looks like this:

TH2F *hTDC2DModule[NR_OF_TDCS];

and the definition/initialization:

   for(int counter=0;counter<TDCModules;counter++){
	  sprintf(name,"hTDC2DModule%d",counter);
	  sprintf(title,"hTDC2DModule %d ",counter);
          hTDC2DModule[counter]=new TH2F(name,title, 4000, -8000, 14999,128,0,128);
   }

(you will notice that I replaced the "H2_BOOK" in previous versions with "new TH2F" 
and I also got rid of the "static" keyword.)

Is this not already what is suggested in the example of TGraphErrors?
The minute I change NR_OF_TDCS with TDCModules the compilation fails as follows:

main.c:261:30: error: array bound is not an integer constant before ‘]’ token
 TH2F *hTDC2DModule[TDCModules];

It seems to me C++ does not allow variable size arrays. Period.
So we are stuck. 
On the other side, this only affects the TDC2DModule histograms. For now we just live with the problem and fix it experiment by experiment?

KevinCWLi, 2016-05-30:
Maybe a double pointer will do?

So you would have something like:

TH2F **hTDCDModule;

Then in the main code you can have something like:

hTDCDModule = new TH2F*[TDCModules];

so now it can be initialised with a dynamic variable (int).

Something like that…

Kind regards
Kevin

padsley, 2016-05-30:
Kevin is, I think, right. I hadn't noticed that at all when I gave you the example ablove. The problem is that I was trying to make use of an example where the TGraph array is defined in the code not outside it.

I've managed to get it working by doing the following:

//static TH2F *hTDC2DModule[NR_OF_TDCS];
TH2F **hTDC2DModule;

**************************************

 hTDC2DModule = new TH2F*[TDCModules];
   
   for(int counter=0;counter<TDCModules;counter++){
	  sprintf(name,"hTDC2DModule%d",counter);
	  sprintf(title,"hTDC2DModule %d ",counter);
          //hTDC2DModule[counter]=H2_BOOK(name,title, 4000, -8000, 14999,128,0,128);
	  hTDC2DModule[counter] = new TH2F(name,title,4000,-8000,14999,128,0,128);
   }

That compiles but I don't know if it runs properly...

padsley, 2016-05-30:
(Mea culpa, mea culpa)

KevinCWLi, 2016-05-30:
Hey Phil

I believe you will have to make it a double pointer declaration:

TH2F **hTDCDModule;

the one pointer will be because it is an object and the second designates it as an array of those objects;

Then you can write:

hTDCDModule = new TH2F*[TDCModules];

and then your usual code should work:

for(int counter=0;counter<TDCModules;counter++){
sprintf(name,"hTDC2DModule%d",counter);
sprintf(title,"hTDC2DModule %d ",counter);
hTDC2DModule[counter] = new TH2F(name,title,4000,-8000,14999,128,0,128);
}

neveling, 2016-05-30:
Hi guys
Thanks for the help.
I did the changes as summarized below. The code compiles and from a brief look at things it seems as if it is working nicely.

Will make the changes to my repository and then request to push them through to this master

in main.c replaced in the histogram definition

static TH2F *hTDC2DModule[NR_OF_TDCS];

and the definition/initialization:

for(int counter=0;counter<TDCModules;counter++){
   sprintf(name,"hTDC2DModule%d",counter);
   sprintf(title,"hTDC2DModule %d ",counter);
   hTDC2DModule[counter]=new H2_BOOK(name,title, 4000, -8000, 14999,128,0,128);
}

with

TH2F **hTDC2DModule;

and
 
hTDC2DModule = new TH2F*[TDCModules];
for(int counter=0;counter<TDCModules;counter++){
    sprintf(name,"hTDC2DModule%d",counter);
    sprintf(title,"hTDC2DModule %d ",counter);
    hTDC2DModule[counter]=new TH2F(name,title, 4000, -8000, 14999,128,0,128);   
}

Similarly in adc.c replaced

static TH2F *hADC2DModule[5];

and 
   
   for(int counter=0;counter<5;counter++){
	  sprintf(name,"hADC2DModule%d",counter);
	  sprintf(title,"hADC2DModule %d ",counter);
          hADC2DModule[counter]=H2_BOOK(name,title,4096,0,4096,32,0,32);
   }

with

TH2F **hADC2DModule;

and

   hADC2DModule = new TH2F*[ADCModules];   
   for(int counter=0;counter<ADCModules;counter++){
	  sprintf(name,"hADC2DModule%d",counter);
	  sprintf(title,"hADC2DModule %d ",counter);
          hADC2DModule[counter]=new TH2F(name,title,4096,0,4096,32,0,32);
   }

padsley, 2016-05-30:
This problem is now fixed. Someone should go through the code in the future to check that the array sizes are all limber and loose and not fixed definitions. Yoga-like arrays, if you will.

closed 2016-05-30 by padsley