padsley/k600analyser #102: Changes to main.c (suggested by jjvz)

jjvz, 2016-05-28:
1)  Add the lines below near line 824 in main.c (this creates the branches containing gammadata stuff):
 #ifdef _GAMMADATA
  gROOT->ProcessLine(".L Parameters.c+");
  gROOT->ProcessLine("#include \"GammaData.h\"");   // may be overkill...?
  gROOT->ProcessLine(".L GammaData.c+");
  t1->Branch("GammaInfo","GammaData",&gammy);
#endif  

2) In the switch in line 1215 in main.c:
switch(channel){
	case 3:  t_k600=1; break;
....
case = 2 and 3 are also used for POLARISATION experiments (line 106 of main.c), so there may be a clash in future if we ever run POL experiments again, so maybe put a error flag if it is duplicated: 
#ifdef _POLARIZATION              // for PR153, Sept/Oct 2010
const int TDC_CHAN_POLU=2;	
const int TDC_CHAN_POLD=3;	 
#endif

padsley, 2016-06-01:
For reference, @neveling said that he'd done point (2) in this issue.

neveling, 2016-06-01:
I moved declaration of TDC_CHAN_POL to Parameters.h and removed the t_k600 field. This was originally a weak attempt to emulate a pattern register. Fact is that we dont need to label an event as a k600 event as all our events are per definition k600 events. But it is handy to label an event a pulser event, which is now the first case condition after the switch command. Also, I changed the numbers of the pol variables to something not used so far.

closed 2016-08-07 by padsley