padsley/k600analyser #120: RAW data: ADC channel values also needed

neveling, 2016-06-19:
I think it would be useful if the ADC channel nrs and module nrs are also given in the raw data.

neveling, 2016-06-19:
What an idiotic thing to say. 
And I am not even on night shift.
Please ignore

neveling, 2016-06-19:
after thinking another few secs, I am confused again

padsley, 2016-06-19:
http://xkcd.com/1695/

neveling, 2016-06-20:
I have now convinced myself that it is a good thing to have ADC channels as a separate array as well. See the attached picture. This is a quick way to see which ADC channels belong to a particular detector. The draw cmd was as follows:

DATA->Draw("ADCValues[60]:ADCChannels>>h(130,0,130,100,300,4000)","ADCValues>700","col")

[image]

padsley, 2016-06-28:
I can't sleep...

Has this been implemented in the master branch? If so, can the issue be closed?

neveling, 2016-06-29:
Indeed it is implemented.
Will close this issue

closed 2016-06-29 by neveling