padsley/k600analyser #123: NaI_vicente branch: Problems with Silicon code

pellegri, 2016-07-12:
while compiling the code with the new Scintillator sort code I had a problem with the silicon code:

Error: SiliconData() declared but not defined :0:
Error in <TClass::BuildRealData>: Cannot find any ShowMembers function for SiliconData!
Error in <TBranchElement::InitializeOffsets>: Could not find the real data member 'SiliconHits' when constructing the branch 'SiliconInfo' [Likely missing ShowMember].
 
then this:

SiliconTheta suggests a silicon detector at forward angles - problem in angle computation

and at the end segmentation fault:

 *** Break *** segmentation violation

===========================================================
There was a crash.
This is the entire stack trace of all threads:
===========================================================

Thread 2 (Thread 0x7f384b9e2700 (LWP 25876)):
#0  0x00007f384eb527ad in accept () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007f38517c97cb in TUnixSystem::AcceptConnection(int) () from /home/luna/root/lib/libCore.so
#2  0x00007f3850d6e7b3 in TServerSocket::Accept(unsigned char) () from /home/luna/root/lib/libNet.so
#3  0x0000000000419d5f in root_socket_server (arg=<optimized out>) at src/mana.cxx:5272
#4  0x00007f384f4a358e in TThread::Function(void*) () from /home/luna/root/lib/libThread.so
#5  0x00007f384eb496fa in start_thread (arg=0x7f384b9e2700) at pthread_create.c:333
#6  0x00007f384e0dfb5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 1 (Thread 0x7f38520a7140 (LWP 25863)):
#0  0x00007f384e0a469b in __GI___waitpid (pid=25942, stat_loc=stat_loc
entry=0x7fffb3c8f040, options=options
entry=0) at ../sysdeps/unix/sysv/linux/waitpid.c:29
#1  0x00007f384e01dfab in do_system (line=<optimized out>) at ../sysdeps/posix/system.c:148
#2  0x00007f38517cb2b4 in TUnixSystem::StackTrace() () from /home/luna/root/lib/libCore.so
#3  0x00007f38517cd55c in TUnixSystem::DispatchSignals(ESignals) () from /home/luna/root/lib/libCore.so
#4  <signal handler called>
#5  0x00000000004460ae in main_eor (run_number=10194) at main.c:1950
#6  0x00000000004202c9 in eor (run_number=10194, error=error
entry=0x7fffb3c91ba0 "\360\004\vR8\177") at src/mana.cxx:1934
#7  0x0000000000421326 in analyze_run (run_number=run_number
entry=10194, input_file_name=input_file_name
entry=0x7fffb3c91df0 "/home/luna/data/PR238/2016/run10194.mid.gz", output_file_name=output_file_name
entry=0x7fffb3c91ef0 "") at src/mana.cxx:4290
#8  0x000000000042175b in loop_runs_offline () at src/mana.cxx:4363
#9  0x0000000000419a9e in main (argc=<optimized out>, argv=<optimized out>) at src/mana.cxx:5579
===========================================================

The lines below might hint at the cause of the crash.
If they do not help you then please submit a bug report at
http://root.cern.ch/bugs. Please post the ENTIRE stack trace
from above as an attachment in addition to anything else
that might help us fixing this issue.
===========================================================
#5  0x00000000004460ae in main_eor (run_number=10194) at main.c:1950
#6  0x00000000004202c9 in eor (run_number=10194, error=error
entry=0x7fffb3c91ba0 "360004vR8177") at src/mana.cxx:1934
#7  0x0000000000421326 in analyze_run (run_number=run_number
entry=10194, input_file_name=input_file_name
entry=0x7fffb3c91df0 "/home/luna/data/PR238/2016/run10194.mid.gz", output_file_name=output_file_name
entry=0x7fffb3c91ef0 "") at src/mana.cxx:4290
#8  0x000000000042175b in loop_runs_offline () at src/mana.cxx:4363
#9  0x0000000000419a9e in main (argc=<optimized out>, argv=<optimized out>) at src/mana.cxx:5579
===========================================================

[Analyzer,ERROR] [midas.c:1952:,ERROR] cm_disconnect_experiment not called at end of program
Cannot open message log file '/home/neveling/K600/SOFTWARE/git/k600analyser-1/midas.log', open() errno: 2 (No such file or directory)

padsley, 2016-07-12:
There are two separate things here. One is that the SiliconData stuff isn't working. The other is the angle computation for the SiliconTheta value. Due to the limited size of the array that @KevinCWLi made for the original angle computation (he only had four detectors), when @neveling added it to the code, it worked fine until there was a fifth detector.

Given that, I commented a large amount of the angle computation out at some point. It's possible that it hasn't been added back in in the present version of the code.

The other problem I'm trying to chase down.

padsley, 2016-07-12:
This error isn't occurring for me.

padsley, 2016-07-12:
A couple of questions:

> while compiling the code with the new Scintillator sort code I had a problem with the silicon code:

While compiling? Or while running? It compiles fine for me.

What command are you using to run the code and with what file? Currently I'm not seeing any error from the file. Looking at the stack trace above, it doesn't appear to reach SiliconData (or anything else for that matter). It looks like it's flaking out at loop_runs_offline().

pellegri, 2016-07-12:
sorry while running. It compiles fine.
I’m using this command line ./analyzer -i run10194.mid.gz

padsley, 2016-07-12:
HHHHHHHHHHHHHHMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM.

I'm stuck. Maybe I should have a beer and not spend my day off playing for C++...

pellegri, 2016-07-12:
Nooooooooo don’t abandon me pleaseeee

padsley, 2016-07-12:
[image]

pellegri, 2016-07-12:
ahahahhaha I will never leave u alone for that ;)

padsley, 2016-07-12:
OK.

I still see no problems.

Can you send me that run so that I can try it with that?

pellegri, 2016-07-12:
Sorry Phil, I'm doing one hundreds of things at the same time.
In attachment you can find the run. Can be something related to the odbedit
file?

​
 run10194.mid.gz
<https://drive.google.com/file/d/0B2mXlkNxhvVfbUJ4enZfbHZNMms/view?usp=drive_web>
​

padsley, 2016-07-12:
I still see no problem.

I've changed the angle computation to always return 180 for the moment because the problem format for this computation isn't working. That still needs to be fixed. Will push change to repo.

pellegri, 2016-07-12:
Ok thanks Phil.
I’ll try to understand which is the problem. 

Cheers,
Luna

padsley, 2016-07-27:
The problem (for reference) was that the libcint.so file hadn't been generated when ROOT was built. Recompiling ROOT fixed it.

closed 2016-07-27 by padsley