padsley/k600analyser #124: Merging the nai_vicente branch with PR238_2016 and master one

pellegri, 2016-07-13:
I added the notification to the gamma sort code. In the branch nai_vicente there are the codes for Hagar, Clover and Scintillator. Can someone merge them with the one now in use?

padsley, 2016-07-13:
You want the branch vicente_nai merged with the master branch?

pellegri, 2016-07-13:
I want first to merged it with the PR238_2016 and second to merge it with the master branch so that the changing on the Gamma codes can be added in both branches

Thanks Phil

padsley, 2016-07-27:
I think I did this. If I didn't, let me know...

pellegri, 2016-07-27:
We merged the PR238_2016 with nai_vicente but we didn’t merged with the master. We will use the master branch for PR264 or we do another branch?

L

padsley, 2016-07-27:
Oh... OK. Should I merge nai_vicente into the master or PR238_2016 into the master?

nichubbard, 2016-07-27:
I made a new branch for PR264, which currently is based on master.

pellegri, 2016-07-27:
We could try to merge the PR238_2016 with the PR264 and see how it looks. If everything goes fine we can merge it with the master…. ok?

L

padsley, 2016-07-27:
OK. I'll try that.

Dr Philip Adsley MA MSci (Cantab)
Postdoctoral Research Fellow, University of Stellenbosch/iThemba LABS

pellegri, 2016-07-27:
Did you do that? or I should do it?

L

padsley, 2016-07-27:
Merged PR238_2016 with PR264. Do you want to try testing it to see how it goes?

pellegri, 2016-07-27:
I’ll do it now now.

L

pellegri, 2016-07-27:
Btw I still have the problem with the root lib on my laptop. I had to
upload them by hand again.

In anycase I run it and it gives me this error:
[Analyzer,ERROR] [mana.cxx:497:db_get_event_definition,ERROR] Cannot find
event id 3 under /equipment
Cannot open message log file
'/home/neveling/K600/SOFTWARE/git/k600analyser-1/midas.log', open() errno:
2 (No such file or directory)

a part from this it works. I'll check again later with the calibration data
that I'm running.

neveling, 2016-07-27:
The error about event id 3 is not important.
It means it has troubles with the beamline EPICS event. This is not crucial to your current analysis so for now I suggest you ignore it. I asked for the beamline event to be switched on again for this weekend.

And about the message about the log file: that can be fixed by changing the directory name in
the odb under /Logger/Data dir/

padsley, 2016-07-27:
@pellegri - what problem with the ROOT libraries? The ROOT compilation itself or the compilation of the analyser?

pellegri, 2016-07-27:
The one that you fix re-compiling root but it didn't work. I had to do again .L GammaData.c+ etc.

pellegri, 2016-07-28:
@neveling which is the command to change the /Logger/Data dir/?