padsley/k600analyser #127: TDCOffsets

padsley, 2016-07-17:
This is currently borked because it tries to set the parameters at the point where things are read which is before 'TDCInit' creates the structure to put it in.

Changed so that it works like the ADC calibration but only in PR244 branch. Needs to be propagated into the master.

padsley, 2016-07-27:
I've fixed this in PR244 but the fix in the master branch is going to wait patiently (expectans expectavi, innit) until after PR264 so that I don't break the master just before the experiment.

padsley, 2016-07-27:
#131

padsley, 2017-02-01:
Uh. Again with the "I think that I did this."

neveling, 2018-03-21:
I came across this issue during a look at PR259. I dont think it is implemented in the master branch yet.
I shall now attempt to do that.

padsley, 2018-03-27:
> I've fixed this in PR244 but the fix in the master branch is going to wait patiently (expectans expectavi, innit) until after PR264 so that I don't break the master just before the experiment.

Or alternatively I could just forget to put the change into the master branch after the experiment. Oops.

neveling, 2018-03-27:
Hehehe (^;
I forgot to say, it is now implemented on the master branch. Lets keep this issue open as I am pretty sure the thing will still crash despite my best intentions.

padsley, 2018-07-13:
Having not seen any complaints about this, I'm calling it fixed. Reopen the issue if you need to.

closed 2018-07-13 by padsley