padsley/k600analyser #135: PR264 Crash due to SetDetectorType()

nichubbard, 2016-07-29:
On my laptop I get segfaults immediately when trying to sort data, which I have now identified as being caused when SetDetectorType() is called in the ScintillatorSort. Commenting out that line fixes the crashes for me and is not a problem for this experiment with one type of Gamma detector, but still. Can someone try to reproduce? If so I will push the change commenting it out (for now), if not I will remain baffled at my laptop.

```
The lines below might hint at the cause of the crash.
If they do not help you then please submit a bug report at
http://root.cern.ch/bugs. Please post the ENTIRE stack trace
from above as an attachment in addition to anything else
that might help us fixing this issue.
===========================================================
#6  0x00007f0f5a0eb69a in strlen () from /lib64/libc.so.6
#7  0x00007f0f5e40d2b5 in TString::TString (this=0x7ffe70181620, cs=0xc <error: Cannot access memory at address 0xc>) at /home/nic/Software/Build/root-6.06.02/core/base/src/TString.cxx:106
#8  0x00007f0f5df99850 in TGenCollectionStreamer::WriteObjects (this=this
entry=0x35a01a0, nElements=nElements
entry=4, b=...) at /home/nic/Software/Build/root-6.06.02/io/io/src/TGenCollectionStreamer.cxx:1006
#9  0x00007f0f5df9c018 in TGenCollectionStreamer::Streamer (this=0x35a01a0, b=...) at /home/nic/Software/Build/root-6.06.02/io/io/src/TGenCollectionStreamer.cxx:1480
#10 0x00007f0f5decd82f in TCollectionStreamer::Streamer (this=0x3591a90, buff=..., pObj=<optimized out>, onFileClass=0x0) at /home/nic/Software/Build/root-6.06.02/io/io/src/TCollectionProxyFactory.cxx:166
#11 0x00007f0f5def5e05 in TClass::Streamer (onfile_class=0x0, b=..., obj=0x534d130, this=0x3584280) at /home/nic/Software/Build/root-build/include/TClass.h:547
#12 TBufferFile::WriteFastArray (this=0x5828d30, start=<optimized out>, cl=0x3584280, n=<optimized out>, streamer=<optimized out>) at /home/nic/Software/Build/root-6.06.02/io/io/src/TBufferFile.cxx:2283
#13 0x00007f0f5dec577b in TStreamerInfo::WriteBufferAux<char**> (this=0x5202d30, b=..., arr=
0x7ffe70181878: 0x7ffe70181870, compinfo=0x5300e88, first=first
entry=0, last=last
entry=1, narr=1, eoffset=0, arrayMode=0) at /home/nic/Software/Build/root-6.06.02/io/io/src/TStreamerInfoWriteBuffer.cxx:590
#14 0x00007f0f5de2d3dd in TStreamerInfoActions::GenericWriteAction (buf=..., addr=<optimized out>, config=<optimized out>) at /home/nic/Software/Build/root-6.06.02/io/io/src/TStreamerInfoActions.cxx:174
#15 0x00007f0f5def4aa5 in TStreamerInfoActions::TConfiguredAction::operator() (this=<optimized out>, this=<optimized out>, object=0x534d0f0, buffer=...) at /home/nic/Software/Build/root-build/include/TStreamerIn
foActions.h:116
#16 TBufferFile::ApplySequence (this=0x5828d30, sequence=..., obj=0x534d0f0) at /home/nic/Software/Build/root-6.06.02/io/io/src/TBufferFile.cxx:3967
#17 0x00007f0f5c9d27aa in TBranch::Fill (this=this
entry=0x5300540) at /home/nic/Software/Build/root-6.06.02/tree/tree/src/TBranch.cxx:758
#18 0x00007f0f5c9b9200 in TBranchElement::Fill (this=0x5300540) at /home/nic/Software/Build/root-6.06.02/tree/tree/src/TBranchElement.cxx:1231
#19 0x00007f0f5c9b8f88 in TBranchElement::Fill (this=0x358bfa0) at /home/nic/Software/Build/root-6.06.02/tree/tree/src/TBranchElement.cxx:1256
#20 0x00007f0f5c98bf01 in TTree::Fill (this=0x370f3d0) at /home/nic/Software/Build/root-6.06.02/tree/tree/src/TTree.cxx:4345
#21 0x00000000004454b7 in main_event (pheader=0x7f0f41c33010, pevent=0x7f0f41c33020) at main.c:1911
#22 0x000000000041db1e in process_event (par=par
entry=0x6a7be0 <analyze_request>, pevent=pevent
entry=0x7f0f41c33010) at src/mana.cxx:3155
#23 0x0000000000420e78 in analyze_run (run_number=run_number
entry=24, input_file_name=input_file_name
entry=0x7ffe70182250 "../data/run00024.mid.gz", output_file_name=output_file_name
entry=0x7ffe70182350 "") at src/mana.cxx:4176
#24 0x00000000004213b1 in loop_runs_offline () at src/mana.cxx:4364
#25 0x0000000000419de2 in main (argc=<optimized out>, argv=<optimized out>) at src/mana.cxx:5577
===========================================================
```

pellegri, 2016-07-29:
mmmmmm this is weird because I'm using the same code an it doesn't crash on
my laptop and desktop. Are you using the PR264 branch?

padsley, 2016-07-29:
> root.cern.ch/bugs. Please post the ENTIRE stack trace

Are you using the same configuration file?

Dr Philip Adsley MA MSci (Cantab)
Postdoctoral Research Fellow, University of Stellenbosch/iThemba LABS

nichubbard, 2016-07-29:
I am using the PR264 branch with no changes to the configuration file.

nichubbard, 2016-07-29:
While running scripts to calibrate the NaI, I have been getting other weird errors and messages from ROOT (although they don't crash), so for now I am willing to assume my ROOT install is messed up somehow. I will update after I can be bothered to rebuild it.

nichubbard, 2016-07-29:
Recompiled ROOT and now it does not crash. I shall not question it.

closed 2016-07-29 by nichubbard

pellegri, 2016-07-29:
Great! Thanks good.
I also did some calibrations. I’m coming down in a minute ;)