padsley/k600analyser #140: Non-linear calibration

padsley, 2016-08-16:
@pellegri has requested implementation of a non-linear calibration.

I am going to think about how to implement this before, y'know, implementing it.

padsley, 2016-08-16:
I have done a simple version of this but I need to make a couple of quick changes:

1: The code must be protected against not having enough ADC channels defined.

2: There must be default values set for the ADC parameters in the case that nothing is otherwise initialised.

pellegri, 2016-08-16:
Thanks Phi!

What do you mean with the 1.?
Yes for the second true. I guess that you should put for E=aCh^2+bCh+c a=0,b=1 and c=0

L

padsley, 2016-08-16:
Uh.

So, point 1 was that I had originally set the ADC calibration value arrays up so that they were the same size as the number of ADC channels, so every channel had a couple of calibration parameters.

For the second point, the structure of the calibration is now:

`ADCCalibrationParameters[channel[0]` = number of calibration parameters that are available.
`ADCCalibrationParameters[channel][1+p]` = calibration parameter of order p.

I.e.:

```
double CalculateEnergy(int channel, int value)
{
double energy = 0;

for(int i=1;i<ADCCalibrationParameters[channel][0]+1;i++)
{
energy += ADCCalibrationParameters[channel][i] * pow(adc_value,i-1);
}
return energy;
}
```

I've implemented this in RawData and CloverSort. It can be implemented in other places as-and-when required.

I've also set the code so that it doesn't change the ADC calibration parameters used elsewhere, but rather sets the parameters for these when it sees a linear calibration. I.e., ADCGains and ADCOffsets are still usable if desired.

padsley, 2016-08-16:
FYI: ADCCalibrationParameters has type double**

It's 32*ADCModules long. Each 'row' is set  by the number of parameters used for that channel calibration. Therefore each row is usually of size '2' but can be larger.

closed 2016-08-16 by padsley