padsley/k600analyser #142: Weird X1pos due to hot wires?

nichubbard, 2016-09-22:
Looking at Run #1085 from PR264 the graph at approx 425mm looks very weird, which *seems* to be due to some hot TDC wires. Phil has had a quick look at the code whilst he's at York and we're stuck. There is a function to remove the hot wires but that doesn't seem to be called - and we're not sure why.

Comments, suggestions?
[image]

nichubbard, 2016-09-22:
[image]

neveling, 2016-09-22:
Mmm, not nice.
Will look at it first thing tomorrow.

nichubbard, 2016-09-22:
The problem seems to have been solved by deleting "<<<<<HEAD" from the first line of CableLength.dat. How it appeared and didn't cause crashes I do not know.

closed 2016-09-22 by nichubbard

padsley, 2016-09-26:
Hi. @nichubbard, could you make sure that the PR264 online version is updated with the right files (i.e. without <<<<<<<<<<<<<<<<<<<<<<<<<<<HEAD included). Just in case someone else ever needs to use that version of the analyser.