padsley/k600analyser #143: Problems running things at York on the NPG machine

padsley, 2016-10-06:
```
[Analyzer,ERROR]` [system.c:719:ss_shm_open,ERROR] Cannot create shared memory segment '/Default_ODB_SHM', shm_open() errno 13 (Permission denied)
[Analyzer,ERROR] [midas.c:2177:cm_connect_experiment1,ERROR] cannot open database

Error: Unexpected error #304
```

padsley, 2016-10-06:
I'm trying to set things up for one of the York students to look at analysis things and I'm getting a variety of error messages, related (I think) to the ODB.

gauss345, 2016-10-06:
do you have permissions? if you running locally, is the user that run the analyzer  the same uid/gid?

nichubbard, 2016-10-06:
This I believe is due to missing environmental variables, and it defaulting to then trying to write files into /, which is not allowed on the York systems.
This is the MIDAS/ROOT section of my .bashrc, I think it is MIDAS_DIR that is missing in this case.

```
export MIDASSYS=/npdisks/data4/njh521/TRIUMF_MIDAS/midas
export MIDAS_DIR=/npdisks/data4/njh521/iThemba-PR228/midas_data
export ROOTSYS=/npdisks/sfw64/root
export PATH=$MIDASSYS/linux/bin:$ROOTSYS/bin:$PATH
. $ROOTSYS/bin/thisroot.sh
```

gauss345, 2016-10-06:
a errno 13 error would normally indicate that you do not have permission to access the directory/file/shm on disk, even if MIDAS_DIR is set.

I assumed here, the correct setup procedure was followed ... :)  But I think Nic might be right...

neveling, 2016-10-07:
Do you get the error when you try to run the analyser, or when you try to open the ODB?

What happens when you open the ODB?

padsley, 2016-10-07:
So, I've installed the programme on the student's laptop instead of on the NPG system.

I have a feeling that @nichubbard was right - it's probably an issue with permissions. The MIDAS directory that I was trying to use is a shared one and this is (probably) what's screwing everything up.

Anyway, for now this isn't a problem.

padsley, 2018-07-13:
I'm not sure that this problem has ever been fixed but the student moved onto other things so I don't think that it's currently a problem.

I'm going to ascribe this to gremlins and close the issue.

closed 2018-07-13 by padsley