padsley/k600analyser #144: Cannot find event id 3 under /equipment

nichubbard, 2016-10-11:
while sorting PR264 runs you get spammed with the following "error"
[Analyzer,ERROR] [mana.cxx:497:db_get_event_definition,ERROR] Cannot find event id 3 under /equipment

My understanding is this isn't really a problematic error, but as I find it a minor nuisance I was wondering if there was a quick change to the ODB/code that could be made to silence it? If not I will just put up with it.

padsley, 2016-10-11:
Uh. Probably?

IIRC, it's the slow control data class. Things like beamline behaviours, magnetic fields etc.

Retief is the person for that.

neveling, 2016-10-11:
Yes, it represents the epics beamline related info. It is defined as event ID 3 in analyzer.c
The problem should be fixed by adding an event 3 in the ODB.  I attach an example ODB file.
Had to change it to .txt to get GITHUB to accept it
[basic.odb.txt](https://github.com/padsley/k600analyser/files/522127/basic.odb.txt)

nichubbard, 2016-10-11:
Thanks, that solved it. I'll push a commit to PR264 with this change included in a bit.

closed 2016-10-11 by nichubbard