padsley/k600analyser #148: FYI | PR236 - issue with histosPR236.C and line colour

LindsD, 2017-02-12:
@Chane13M has been working through the user manual and got to the point where she needed to plot two spectra on the same axes in different colours. This is what was put into root:

```
DATA->Draw("X1pos->hX1pos");
hX2pos->SetLineColor(6);
DATA->Draw("X1pos->hX2pos",CUTbasic,"same");
```

The two graphs would be drawn on the same axes but no matter what was attempted, the line colours would not change. This also didn't seem to be a well-documented issue online. Eventually, I decided to open the histosPR236.C file and take a look. I added the following lines to the beginning of the script:

```
#include <TH1.h>
#include <TH2.h>
#include <TCutG.h>

void histosPR236()
```

This fixed the problem completely. Just wanted to let you know in case other students have the same issue. Did you come across this when you used histosPR236.C??

padsley, 2017-02-12:
I'm debating closing this issue because I don't think that it's a k600analyser problem. I think that it's either user error or a problem with ROOT.

The commands as written should draw the second histogram on the same plot. That seems legit. The setting of the line colour should simply be done using:

Histo2->SetLineColor(2);

Note that ROOT doesn't always update a histogram if you're not interacting with that histogram. I.e., you may have to click on the frame containing the histogram to see the canvas update. You can also call TCanvas::Update() if you want to form the canvas to repaint.

padsley, 2017-02-12:
Having just looked again: I'm fairly sure that you have to create the histogram and fill it before setting the line colour. I think ROOT sets a default line colour when the TTree::Draw() command is invoked. I think the ordering of the commands may be faulty.

ggonei, 2017-02-12:
Phil; you'd be right, except the object exists already so you can manipulate linecolor property whenever, as it is part of the TH1 object.

TH1 and TH2 don't need including.  TCutG is the same as the cut class but has graphical manipulation...stuff.

When the TCutG object is created by TTree::Draw, it is added to the list of special objects in the main TROOT object pointed by gROOT.  The Graphical cut can be edited like a normal TGraph.  The draw command succeeds with the cut because of TCut basic, but this destroys any TGraph information (and thus the function SetLineColor).

Basically, I'm pretty sure applying the cut makes a new object which is not manipulated with the call hX2pos->SetLineColor(6);

padsley, 2017-02-12:
@ggonei: I realised afterwards that I don't do histograms like this, which was why my TTree::Draw was always changing the line colour.

I think that part of the problem is that the TTree::Draw should be:

`DATA->Draw("varexp>>histo");`

not

`DATA->Draw("varexp->histo");`

If I change that one of my computer, then I stop getting spectra filled by the TTree and get the defaul histograms instead.

ggonei, 2017-02-23:
That looks right.  >> pushes data from varexp to histo, -> tries to do pointer logic using varexp and histo.

The fix anyway = include TCutG.h

closed 2017-03-03 by padsley