padsley/k600analyser #149: chi squared versus r-squared

neveling, 2017-03-03:
The naming convention should change: in the code the raytracing provides a goodness of fit parameter that is called chi-squared. This is inaccurate as it is actually a r-squared value.
The naming convention has to change, as it is confusing collaborators.
I note it here as an issue as I dont have time to do it now but will get to it in future, and at least now it is documented

padsley, 2017-03-03:
At the risk of asking a stupid question...

Isn't the only difference a multiplicative factor which is the position resolution of each wire hit? Or is it more complex?

Also, I know that some things like gamma tracking use "Figure of Merit" to refer to a numerical quantity denoting the "merit" of a result that doesn't have a well-defined statistical behaviour so that may be an option for a consistent naming with other codes.

LindsD, 2017-03-04:
I think @neveling is correct. When one plots the "Chi-squared: X1" spectrum, it is centred around 1 with a tail towards zero (characteristic of R^2 since R^2 is always between 0 and 1 with a value of 1 indicating that the fit explains all the variability of the response data around its mean). If someone plots this spectrum thinking that it is chi-squared then they'd assume something is horribly wrong since they'd expect a spectrum centred around 0 with a tail towards 1. 

Now I'm not sure if this is a stupid question but can we not just change all instances of "chi" to "R" in the applicable files? And add a comment that we are using a "goodness of fit" method?