padsley/k600analyser #150: Merge of PR251 with Master branch

pellegri, 2017-03-05:
The branch PR251 contains all the recent modifications regarding the gamma branch and the pad1 offset correction. Can we merge PR251 branch with the master branch? Thanks ;)

padsley, 2017-03-06:
I've done the initial merge but I need @neveling to look into a compilation issue.

Otherwise everything is done (I think).

padsley, 2017-03-06:
experim.h and basic.odb both haven't been saved from merge conflicts as of yet.

They probably should be replaced with updated files. Once that's done, tell me and I'll fettle everything back across for the PR251 and master branches.

pellegri, 2017-03-07:
Phil, I made another change in the main.c in the PR251 since there was an error in the X1offsets part. Can I push the changes in the PR251 branch or should I wait? Actually these changes should be implemented in the master branch too

padsley, 2017-03-07:
Can you wait? I need Retief to fix some other problems first. I don't understand what he did with the structs in experim.h and I don't want to mess things up...

pellegri, 2017-03-07:
Actually I just noticed that you can leave it like that. It's just that the offsets for the X1pos will be inconsistent with the TOF and Pad1. In the first case the offset is subtracted and in the other two is added. I actually don't know if there is a reason for the difference... Maybe retief knows it @neveling

neveling, 2017-03-10:
Hi. I made some odb changes to the master branch, and it compiles now. 
I also look at the the latest on the PR251 branch, and it compiles without need for my intervention.
I will spend a bit more time this afternoon to see what it is that I missed in the past few days.

neveling, 2017-03-10:
Made some small changes related to the si pointer: where it is created and deleted. Can you see if you are happy Phil?

padsley, 2017-03-10:
Sure. This was the change that @vpesudo discussed in his e-mail of 8th March?

> Hi guys, 
> 
> The main cause of the drag of the analyser when including silicons data was, in my case, the missing reinicialization of the 'si' pointer within the main_event routine. Basically, I found the we had
> 
> `#ifdef _GAMMADATA`
> `gammy = new GammaData();`
> `#endif`
>    
> but we didn't had an analogous 
> 
> `#ifdef _SILICONDATA`
> `si = new SiliconData();`
> `#endif`

neveling, 2017-03-10:
Yes.

padsley, 2017-03-10:
Then I'm happy. We might just need to check that PR251 and master haven't diverged too much before moving on but I think everything is done.

Does @LindsD need to look at PR217 and master now? :P