padsley/k600analyser #158: Ex

padsley, 2017-03-28:
I've been getting this question a lot so this is a form of broadcast message:

Ex computed by the analyser. You should not trust this thing. When I wrote the code, I think I only implemented it for inelastic scattering (so it definitely is bad for (p,t) and (3He,t)). In addition, part of the way through I realised that I didn't need or want it and so stopped using it. How complete and working it is is therefore an open question.

Now, it might have been fixed by someone else and be all fine. However, unless you actually check that explicitly, then do not trust it.

FWIW, this is something that should probably be fixed at some point and I shall have a look into it when I am able to do so (late 2020 is my next free time).

pellegri, 2017-03-28:
Following this I check the Ex calculator and fix it on the PR251 branch. It should be generalised for all cases but if another coupled of eyes can double check it, it would be better…
I’ll put it on github late today ;)

Luna

padsley, 2017-03-28:
Awesomesauce. Thanks for doing that.

I suggest that, as @LindsD was one of the people who was looking to use the Ex class, that she does the checking for it. (Sorry... for dropping you in it.)

neveling, 2017-03-28:
I have used CalcEx before for inelastic scattering, and I was under the impression it is working
fine for inelastic scattering.

Luna, am I correct that the only changes you did in your commit this afternoon
"046591a fix the CalcEx function in FocalPlane.c"
was the change the division of rigidity by another 1000 to align things with the SPANC output,
and to add Q?

pellegri, 2017-03-28:
At the begging I wrote another function to calculate it and I compare the results with the one that Phil wrote and they were the same. So then I left the Phil one active and I double check the calculation and generalise for other cases than inelastic scattering. Although I don't have data to check if this part is right
so maybe it's easier if someone that is analysing that kind of data can check that.
I've also changed as you noticed the conversion factor for the QBrho.

For the kinematic calculations I made some notes. I will send them to you tomorrow.

Cheers,
Luna

pellegri, 2017-04-12:
@neveling I just realized that the kinematics calculator is implemented on PR251 branch only. We should merged these changes in the master branch as well. Not sure how to do it but I could try...

padsley, 2017-04-12:
Merge that sh*t.

neveling, 2017-04-12:
I see there is still an open pull request on PR251??

pellegri, 2017-04-12:
Done :P

closed 2019-03-08 by padsley