padsley/k600analyser #161: Not getting any wireplane events

padsley, 2018-01-15:
This isn't actually an issue, it's rather a note about a problem that I found and have now fixed fairly easily so that in future it's there as a reference for people.

If you are running the analyser and not seeing any events produced then check the following:

In odbedit, look at the parameters for the drift-time lower and upper limits and compare to the hTDC2DModule histograms thrown out by analyser.

In the version of the ODB that I had, I found that my limits for the drift-times excluded the region where the hits actually were. I assume that this was either due to different beam energies resulting in different flight times or a change in a delay box somewhere but whatever caused it, it meant that I had an empty U1 problem.

This problem was fixed due to my most excellent and brilliant skills of setting the times to be approximately right. :+1: 

```
user@machine:~/codes/k600analyser$ odbedit
[local:Default:S]/>cd Analyzer/Parameters/main/
[local:Default:S]main>ls
x1_driftt_low                   6100
x1_driftt_hi                    8050
x2_driftt_low                   6100
u1_driftt_low                   2600
u2_driftt_low                   6100
x2_driftt_hi                    8050
u2_driftt_hi                    8050
u1_driftt_hi                    5000
lowtof                          0
hitof                           5000
lowpad1                         0
lowpad2                         0
hipad1                          4096
hipad2                          4096
[local:Default:S]main>set u1
u1_driftt_low
u1_driftt_hi
[local:Default:S]main>set u1_driftt_low 6100
[local:Default:S]main>set u1
u1_driftt_low
u1_driftt_hi
[local:Default:S]main>set u1_driftt_hi 8050
[local:Default:S]main>exit
```

closed 2018-01-15 by padsley

neveling, 2018-03-08:
This problem will be seen for anybody analyzing pre-PR251 data with the analyser for PR251 and onwards, and vica versa.

The reason is this: prior to experiment PR251 we did not care about the exact placement of the
drifttime spectrum within the TDC time window, as long as it was all in. We never changed it, so it was always in the range approx 6000ish-8000ish in TDC channel number.

However, in PR251 we wanted to be sure we have a proper coincidence time window for K600 + BaGeL events, so we changed the delay in the GDG going to the TDC trigger inputs, and ch0 of each TDC,
(see the picture of the trigger electronics) to allow more random coinc peaks on the right of the
prompt coincidence time peak (which with the way we do the trigger meant we decrease the delay in the GDG).
This had the effect of moving the drifttimes towards the TDC channel range (3000ish-5000ish).

All post PR251 experiments should have their times in this region (touch wood, you never know who
messes with the electronics so I am not making any promises)

[k600-trigger-electronics.pdf](https://github.com/padsley/k600analyser/files/1793059/k600-trigger-electronics.pdf)

padsley, 2018-03-08:
OK. That seems like a reasonable explanation. One request: what date is the PR251 experiment? October 2016? If we have the date listed here then it's easier for people to check if the data that they're analysing is post-PR251 or pre-PR251.

neveling, 2018-03-08:
Good point.

pre-PR251 means everything up to Oct 2016.

Here is a list of experiments divided in pre and post eras. IT MAY NOT BE 100% ACCURATE SO ALWAYS CHECK WHEN N DOUBT

Pre-PR251 experiments:
PR137, PR153, PR164, PR166, PR167, PR170, PR183, PR184, PR185, PR193, PR194, PR195, PR196, PR203, PR210, PR211, PR217, PR226, PR227 PR228a&b, PR231, PR232, PR236, PR238,PR242, PR244, PR249, PR254, PR259,

PR251 and post-PR251 experiments include:
PR240, PR251, PR260, PR262, PR263, PR264, PR275,

pellegri, 2018-03-08:
It sounds like pre and post apocalypse. 

(Sorry as alway silly comment :P)

L

padsley, 2018-03-08:
Antediluvian? Literally meaning "before the [Noah's] flood"?

pellegri, 2018-03-08:
Yep, you studied :P

L

padsley, 2018-03-08:
I quite like the sound of an "antediluvian experiment". I decree that this
is how they shall be known from henceforth.

Dr Philip Adsley MA MSci (Cantab)
Postdoctoral Research Fellow, CNRS, Orsay