padsley/k600analyser #162: Scaler CII Plot

padsley, 2018-04-24:
I noticed while doing some data analysis that (at least in a version of the code that I was running), the CII data in the tree are wrong (mostly 0, some 2 and 4, nothing higher) and the histogram for CII isn't being filled with many events.

This was PR244 data analysed on my local computer.

Are there known issues about this? If not, we should probably check to find out why the CII values are off. The CIU values are fine FWIW, and the livetime is 100% (it was (a,a') with a light target and for the other similar targets that I've analysed before it's 100% so I'm quite happy with the scaler code) so I can just use CIU within errors.

neveling, 2018-04-25:
The uninhibited trigger, pulser, CI and clock are the first 4 scaler values read from the scaler bank (in that order).  The inhibited versions of these are later in the scaler bank, either from data word 16, or 20, or 32... depending on the number of scaler input channels read out in a particular experiment.
From 2017 onwards I think we had a standard of reading out all 32 chan of the first scaler, so that the inhibited trigger, pulser,  etc always should start at 32 (the first 0-31 being allocated to all things uninhibited). I will check what was the status for PR244 when I can get my hands on midas files tomorrow.

padsley, 2018-07-13:
Is this fixed?

neveling, 2018-07-27:
Yes. My bad, should have closed the issue.

closed 2018-07-27 by neveling