padsley/k600analyser #168: Large files in the Github repository

padsley, 2019-01-17:
There are large files in the repository again. I've had a look and I see:

f2041a145b7e  1.0MiB midas.log
29784836fd61  1.1MiB .ODB.SHM
3f25a91e1df5  1.1MiB .ODB.SHM
5b1a60ee39dc  1.1MiB .ODB.SHM
6874575be798  1.1MiB .ODB.SHM
6ae8efb1a86f  1.1MiB .ODB.SHM
7b897060d604  1.1MiB .ODB.SHM
a048f99cc27a  1.1MiB .ODB.SHM
a20daa87e6ed  1.1MiB .ODB.SHM
a5fb95f5906e  1.1MiB .ODB.SHM
cf0a48a09735  1.1MiB .ODB.SHM
cf2a22f6de7e  1.1MiB .ODB.SHM
dad31abf390b  1.1MiB .ODB.SHM
e725d531de10  1.1MiB .ODB.SHM
ec31f2b34625  1.1MiB .ODB.SHM
faa0ea6e0f08  1.1MiB .ODB.SHM
98ab64f12084  1.7MiB analyzer
c49d7f4b4328  1.7MiB TDCarrays.root
14217fbf0ae5  1.7MiB struck/run00575.mid.gz
e0ca25c0caf7  1.8MiB analyzer
5bce54479322  1.8MiB analyzer
229572c4e6cc  1.8MiB analyzer
7256f4d1e98f  1.9MiB analyzer
9795acd0ba42  2.0MiB run00626.mid.gz
27f0102004e6  2.2MiB analyzer
e9be62827d8b  2.2MiB analyzer
55739e6837dd  2.2MiB analyzer
914e8318c577  2.2MiB analyzer
311dbd0bbc05  2.2MiB struck/libAnalyzer/analyzer_example.exe
113abb2d99a5  2.3MiB analyzer
0f5536eb5c9f  2.3MiB analyzer
b2dc08bd36b2  2.3MiB analyzer
77a75f98f8fd  2.4MiB analyzer
5d1dc1ea95d7  3.3MiB AutoTrim.root
69f15ae2e0a9  3.6MiB struck/libAnalyzer/libAnalyzer.a
99cf3c6ad04d  4.2MiB AutoTrim.root
8c37c208f862  9.2MiB AutoTrim.root
60993e3fed3c  9.7MiB AutoTrim.root
1f2d46abc1d3   11MiB AutoTrim.root
40525a53e5fc   12MiB output.root
9c398df1c8a7   13MiB run00624.mid.gz
24f60a8b7c62   29MiB AutoTrim.root
1e298a4fdda8   39MiB test.root

Lots of ROOT files that should be removed. I should do this at some point soon.

neveling, 2019-01-18:
silly question: can you see in which branches these are?

padsley, 2019-01-18:
You can but I can't remember how one does that yet...

Dr Philip Adsley MA MSci (Cantab)

padsley, 2019-02-09:
I've been trying to remove these things but I can't seem to get rid of them, at least in my local repository. Going to try re-cloning to see if the files have gone from the remote.

gauss345, 2019-02-09:
you should be able to do a git rebase -i and push -f the changes. that should allow you to rm files that was committed erroneously...

padsley, 2019-02-09:
I've decided to put it off until I've had the chance to do the various
updates that I need to do e.g. HPGe addback. Easier to do it afterwards I
think.

Dr Philip Adsley MA MSci (Cantab)

padsley, 2020-03-26:
Meh. I will probably never do it.

closed 2020-03-26 by padsley